ARMmbed/mbed-os #3756: USBHID on STM32 targets hangs if report length is greater than 60

Ingramz, 2017-02-13:
Using [current master branch](https://github.com/ARMmbed/mbed-os/tree/aa6d673d703a099b36d21ad879de7e28e9681cb4) with NUCLEO-F446RE (`USB_STM_HAL` defined manually), the device hangs with the following code once the report length reaches 61. I am guessing this board is not the only STM32 target that exhibits such behavior.

```cpp
#include "mbed.h"
#include "USBHID.h"

USBHID hid(64, 64);

HID_REPORT send_report;

DigitalOut l1(LED1);

int main(void) {
    int report_length = 0;
    l1 = 1;
    while (1) {
        wait_ms(100);

        send_report.length = report_length++;

        if (report_length > 64) report_length = 0;

        for (uint8_t i = 0; i < send_report.length; i++)
            send_report.data[i] = rand() & 0xff;

        hid.send(&send_report);

        l1 = !l1;
    }
}
```

Same code on LPC1768 works as intended and reaches 64.

0xc0170, 2017-02-13:
cc @bcostm @adustm @LMESTM @jeromecoutant

adustm, 2017-02-13:
Hello @Ingramz,
Thanks for reporting the issue. I just submitted PR #3760 to fix it.
NUCLEO_F446RE is not supported by default, to I tried with NUCLEO_F446ZE.
It should fix your issue. 

Kind regards

Ingramz, 2017-02-13:
@adustm thank you for providing a fix so quickly.

I wanted to ask, is there any specific reason why NUCLEO_F446RE is not supported by default? Seeing that it is extremely similar to NUCLEO_F446ZE, it should behave identically.

adustm, 2017-02-13:
You're welcome.

We did not have time to work on every platforms to add USB DEVICE / USB HOST features.
We have then focused on ST platforms that actually have a USB plug on it (mainly NUCLEO with 144 pins and DISCOVERY boards). NUCLEO with 64 pins don't have USB even if the microcontroller can support it.

If by any chance your modifications for NUCLEO_F446RE don't break the code of other platforms, feel free to submit a pull request, we will review it.

Kind regards

Ingramz, 2017-02-13:
Excellent. I will submit a pull request once #3738 gets merged.

commit 094fab43 (2017-02-21): Merge pull request #3760 from adustm/usb_txsize_fix

Fix #3756 for 64 bytes transfers
closed 2017-02-21 by sg-