ARMmbed/mbed-os #3763: Remove/reduce allocation of heap and stack from scatter files for NXP platforms because RTX handles this already

maclobdell, 2017-02-13:
### Description
- Type: Enhancement
- Related issue: n/a
- Priority: Major for some platforms w/ less RAM, minor for others with larger RAM

----------------------------------------------------------------
## Enhancement

mbed OS allocates the heap and stack areas at runtime for ARM compiler and GCC_ARM compiler.  This is done in the file RTX_CM_lib.h.  The heap is allocated at the end of the last data section.  It uses symbols defined in the scatter files such as Image$$RW_IRAM1$$ZI$$Limit for ARM and __end__ for GCC.

Thus, the scatter files do not need to allocate their own heap and stack sections for ARM compiler and GCC compiler.  Although the RW_IRAM1 and __end__ symbols ignore the heap and stack areas (the size of heap and stack is subtracted to determine the start of IRAM1, etc, thus there is no real wasted space), defining the heap and stack in the scatter files limit the size of the available RAM.

For example, reserving 1/4 and 1/8 for heap and stack, reserves quite a lot of space that can't be used for data.  

I recommend of of two options:  

Option 1) Remove the defines for heap and stack completely.   Add a comment that indicates the heap and stack are allocated by RTX at runtime.  Pro:  clean and tidy, data section can grow bigger.  Con: data section could grow too big and not enough heap area reserved.   

Option 2) Reduce the heap and stack down to the amount that mbed OS applications typically actually need at a minimum.  This would serve as the floor.   Pro: more space available for data section to grow bigger.  Con: a minimum amount of stack and heap space is ensured to be kept available.

/* Heap minimum 0x4000 of ram and stack minimum 0x1000 */
#define __stack_size__       0x1000
#define __heap_size__        0x4000

0xc0170, 2017-02-14:
cc @mmahadevan108

mmahadevan108, 2017-03-23:
cc @maclobdell For the ARM compiler, the Stack_Size and Heap_Size will reduce RAM space assigned for the data section. I don't see the RAM getting wasted due to these defines. 

In the case of GCC_ARM, these defines are needed for uvisor. We can reduce it for K66 and other platforms but I don't think it should be removed.

mmahadevan108, 2017-03-23:
cc @maclobdell I will see if I can delete this for GCC_ARM in KL27 since this has limited RAM space. On other platforms I will reduce the size.

closed 2017-07-24 by maclobdell