ARMmbed/mbed-os #3768: Export malfunction of mbed online compiler !

mbedNoobNinja, 2017-02-14:
Note: Why in the mbed online compiler The export function is not working, even there is such functionality added with PR: #3632?

Note: Why there is no information about used RAM and ROM in the `Program Details (Build tab)` when a project is compiled with mbed online compiler?

### Description
- Type: Bug | Question
- Related issue: `#3632`

---------------------------------------------------------------
## Bug

**Target**
VK_RZ_A1H

**Toolchain:**
ARM (mbed online compiler)

**Toolchain version:**
(mbed online compiler)

**Expected behavior**
At least IAR, E2studio & DS5 to become active in the drop down list box when choosing Export program from the mbed online compiler

**Actual behavior**
Only GCC (Code Sourcery) is active and even though it is selected the project still can not be exported

**Steps to reproduce**
Just try with the standard mbed_blinky project

-----------------------------------------------------------------

## Question

**What needs to be done, so used RAM & ROM to become available in the Build tab of Program Details window of the mbed online compiler?**

mbedNoobNinja, 2017-02-14:
Is there any chance to alter the options of the mbed online compiler for example to turn of the compressor
(`--datacompressor=off`) because when compiled for RAM execution this option is on by default and the produced bins are not working.

0xc0170, 2017-02-14:
cc @theotherjimmy

theotherjimmy, 2017-02-14:
Hey @mbedNoobNinja thanks for the patch earlier. The online tools are taken from a released version of mbed. since we just made the release containing the patch you are referring to, it may take a few days for the website to be updated.

theotherjimmy, 2017-02-15:
@mbedNoobNinja Could you open the second issue (`--datacompresson=off`) as another issue? We would like to track that feature request separate from this bug report.

mbedNoobNinja, 2017-02-17:
No need, I solved the problem with separate linker script file, specifying section attribute `NOCOMPRESS` for the section containing the RW, which actually do the trick and that turned the datacompressor off. just that I wanted.

mbedNoobNinja, 2017-09-08:
.... a any **Update** 

> The online tools are taken from a released version of mbed. since we just made the release containing the patch you are referring to, it may take a few days for the website to be updated.

so many releases have been produced so far and the projects still can not be exported for **VK_RZ_A1H** platform?
At least the **IAR** exporter should be working, because offline it works flawlessly. For **E2studio & DS5** I found the reason - forgotten edit of the **py** files `__init__.py`

0xc0170, 2018-06-18:
Tested locally, this seems to be resolved

closed 2018-06-18 by 0xc0170