ARMmbed/mbed-os #3776: Broken STM32F429 Discovery project export (missing files)

vargham, 2017-02-15:
I have a STM32F429 Discovery project and it comiles fine online. I tried to export it tt uVision and Embitz, but the exporter misses some device specific files.

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32F4

**Toolchain:**
Online exporter

**meed-os sha:**
Online IDE with mbed-os lib 5.3.5 (lib rev 136)

**Expected behavior**
Exported project should be compiled offline without errors.

**Actual behavior**
Build error because of missing files from the exported project. The mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI folder is completly missing.

**Steps to reproduce**
Select STM32F429 Discovery target in online IDE, export to offline and try to compile.

0xc0170, 2017-02-15:
cc @theotherjimmy

theotherjimmy, 2017-02-15:
This is a duplicate of https://github.com/ARMmbed/mbed-os/issues/3728

I'll explain:

In both exports one thing changed: `targets.json`. In both scenarios, the online exports were using an old version (5.3.1) of `targets.json`. That old version does not define `MCUXpersso` in #3728's case and `STM32F429xl` in this issue's case. They're actually the same bug.

theotherjimmy, 2017-02-15:
Thanks @vargham for providing all of that info. That helps me with debugging, or in this case, marking as a duplicate.

vargham, 2017-02-21:
Thanks. Tried with uVision5. Same problem, missing files from exported project.

adustm, 2017-05-16:
Hello,
As this is a dupplicate of #3728 and #3728 is closed, could someone close this issue please ?
Kind regards

theotherjimmy, 2017-05-16:
Hi @adustm. I agree, if #3728 is closed this should be closed too.

closed 2017-05-16 by theotherjimmy