ARMmbed/mbed-os #3777: Can not build exported STM32F4 project in EmBitz: undefined reference to `__stack'

vargham, 2017-02-15:
I have a STM32F429 Discovery project and it compiles fine online. I tried to export it to offline, but the exporter misses some device specific files. See this problem in the related issue. After download missing files manually, the project compiles, but at the end there is a linker? error:
```
Creating Hex-file.
c:/program files (x86)/embitz/1.11/share/em_armgcc/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/softfp/crt0.o: In function \`_start':
(.text+0x58): undefined reference to `__stack'
collect2.exe: error: ld returned 1 exit status
```
This project worked fine when exported with previous version (mbed-os lib 5.3.4 (lib rev 135))
### Description
- Type: Bug
- Related issue: `#3776`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4.1

**meed-os sha:**
Exported from online IDE with mbed-os lib 5.3.5 (lib rev 136)

**Expected behavior**
Exported project should be compiled offline without errors.

**Actual behavior**
Build fails.

**Steps to reproduce**
Select STM32F429 Discovery target in online IDE, export to Embitz offline and try to compile.

0xc0170, 2017-02-15:
cc @theotherjimmy @bcostm @adustm @LMESTM @jeromecoutant

theotherjimmy, 2017-02-15:
@vargham That particular variable is defined in the [linker file](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_GCC_ARM/STM32F429xI.ld#L46) and as you pointed out in https://github.com/ARMmbed/mbed-os/issues/3776, that file is not being included in the exported project. As such, this is really the same bug. Thanks again for the detailed bug report, it really helps.

vargham, 2017-02-16:
Thanks. I included the linker script and build working fine now.

nelfadi, 2017-04-30:
please close the ticket when resolved

closed 2017-05-01 by vargham

BoumaizaWajdi, 2018-03-03:
[](url)
[](ur
[image]
)
Hello,
someone could hepl me please ? Thanks .

nelfadi, 2018-03-03:
please search for _sidata  in your project and copy/paste the line here.