ARMmbed/mbed-os #3785: Can not Export LPCXpresso IDE from Online.

Kazu-zamasu, 2017-02-16:
### Description
- Type: Question
- Related issue: `unknown`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NXP1768,824,1549,etc(LPCXpresso IDE)

**Toolchain:**
LPCXpresso 

## Question

**Can not export from online compiler to LPCXpresso.
This issue is all export target of LPCXpresso. please any advice to me.**

theotherjimmy, 2017-02-16:
Hey @Kazu-zamasu Have you tried offline? the error would look something like `toolchain GCC_CR is not supported by profile <redacted>/src/arm-mbed/mbed-os-example-blinky/mbed-os/tools/profiles/debug.json`

I'm going to make a patch to remove `GCC_CR` for good. hold on real quick.

theotherjimmy, 2017-02-16:
see https://github.com/ARMmbed/mbed-os/pull/3791

Kazu-zamasu, 2017-02-16:
@theotherjimmy Thank you for your quickly support!!!  I  will waiting PR marge.

theotherjimmy, 2017-02-16:
Be aware that the online compiler will lag behind this repo

closed 2017-02-17 by Kazu-zamasu