ARMmbed/mbed-os #3787: Rebuild fails if --build option specified

SeppoTakalo, 2017-02-16:
If `mbed build` command have been issued earlier with `--build` parameter, rebuilding of application fails.

Various errors happens. Sometimes it tries to build with some random target, ignoring `.mbed` file. Sometimes it fails to resolve options specified in `mbed_app.json`

## Steps to reproduce
### Build failure
```
$ git clone https://github.com/ARMmbed/example-mbedos-blinky.git
$ cd example-mbedos-blinky
$ mbed deploy --procotol ssh
$ mbed target K64F
$ mbed toolchain GCC_ARM
$ mbed compile --build out/K64F
...
Image: out/K64F/mbed-os-example-blinky.bin
$ mbed compile --build out/K64F
...
/Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: section .flash_config loaded at [0000000000000400,000000000000041f] overlaps section .interrupts loaded at [0000000000000000,00000000000007ff]
/Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: section .text loaded at [0000000000000410,000000000000c3a7] overlaps section .flash_config loaded at [0000000000000400,000000000000041f]
/Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `m_interrupts' overflowed by 1024 bytes
/Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `m_flash_config' overflowed by 16 bytes
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/septak01/src/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build out/K64F" in "/Users/septak01/src/mbed-os-example-blinky"

$ ls BUILD/
K64F              UBLOX_EVK_ODIN_W2
$ cat .mbed
ROOT=.
TARGET=K64F
TOOLCHAIN=GCC_ARM
```

Build fails, and looks like it tried to build Ublox, even that I asked K64F

### Configuration error
```
$ git clone https://github.com/ARMmbed/mbed-os-example-mesh-minimal.git
$ cd mbed-os-example-mesh-minimal
$ mbed deploy --procotol ssh
$ mbed target K64F
$ mbed toolchain GCC_ARM
$ mbed compile --build out/K64F
...
Image: out/K64F/mbed-os-example-mesh-minimal.bin
$ mbed compile --build out/K64F
...
Scan: env
[ERROR] Attempt to override undefined parameter 'mbed-mesh-api.6lowpan-nd-channel-page' in 'application[*]'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build out/K64F" in "/Users/septak01/src/mbed-os-example-mesh-minimal"
```

## Possible problem
Looks like `mbed compile` creates a `.mbedignore` file into `BUILD` folder, but does not create it into a folder specified by `--build` option

```
$ ls -a BUILD
.           ..          .mbedignore
$ ls -a out/
.    ..   K64F
```

Therefore build travels into these folders, causing problems.

Reported originally in https://github.com/ARMmbed/mbed-cli/issues/437

bridadan, 2017-02-16:
cc @theotherjimmy

theotherjimmy, 2017-04-25:
Fixed by #3852

closed 2017-04-25 by theotherjimmy