ARMmbed/mbed-os #3788: STM32F4 online builded bin not working

vargham, 2017-02-16:
None of my STM32F4 projects works with mbed lib 136. I also tried all the official demos of STM32F429 Discovery (Gyro, LCD, LED, Button, etc), but none of them worked. The code compiles with the online IDE, but the MCU does nothing after upload. I tried to switch back to mbed lib rev 135, but the error is remained.
Other, locally builded programs are working fine with the same board.

### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
STM32F4

**Toolchain:**
Online IDE

**Toolchain version:**
Current online IDE with mbed-lib rev 136

**Steps to reproduce**
Create a simple blinky for STM32F429 Discovery target. Itt will compile but won't work.

0xc0170, 2017-02-20:
cc @bcostm @adustm @LMESTM @jeromecoutant @theotherjimmy

jeromecoutant, 2017-02-21:
Hi
There is an issue with DISCO_F429ZI since rev 135
Patch is available: https://github.com/ARMmbed/mbed-os/pull/3716
Jerome

adustm, 2017-05-05:
Hello @0xc0170 
I just tried the mbed-os-example-blinky with NUCLEO-F429ZI in the online compiler. It builds and the program works fine.
Can we close this issue, please ?

Kind regards
Armelle

closed 2017-05-05 by vargham