ARMmbed/mbed-os #3792: STM32 incoming CAN message wrong type and format

vargham, 2017-02-17:
Reading CAN messages with STM32 targets results out of range format and type with extended format (4 instead of 1), and remote type (2 instead of 1). Tried with STM32F302 and STM32F429, same result. I tried also with LPC1768 and the result is OK (1 for extended and 1 for remote).
CANFormat and CANType enums are declared in can_helper.h
```
enum CANFormat {
    CANStandard = 0,
    CANExtended = 1,
    CANAny = 2
};
typedef enum CANFormat CANFormat;

enum CANType {
    CANData   = 0,
    CANRemote = 1
};
typedef enum CANType CANType;
```
Tried:
`printf("CAN in Format=%d Type=%d\r\n",canMessage.format, canMessage.type);`
Output is Format=0 for CAN standard and Format=4 for CAN extended. Output is Type=0 for data and Type=2 for remote. Extended format should be 1 and not 4, remote frame should be 1 not 2.

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32 all

**Toolchain:**
Online IDE

**Toolchain version:**
Current Online IDE with mbed-lib rev 136.

**Expected behavior**
Extended format should be CANExtended (1), remote frame should be CANRemote (1).

**Actual behavior**
Extended format is 4, remote frame is 2.

**Steps to reproduce**
Receive CAN messages (Extended/Standard and Data/Remote) with any STM32 target.

0xc0170, 2017-02-20:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-02-21:
Hi
I think I understood the issue.
In can_api.c file, you can try:
    msg->format = (CANFormat)(((uint8_t)0x04 & can->sFIFOMailBox[handle].RIR)>>2);
    msg->type = (CANType)(((uint8_t)0x02 & can->sFIFOMailBox[handle].RIR)>>1);

I will push a PR when https://github.com/ARMmbed/mbed-os/pull/3737 is merged

Thx
Jerome

closed 2017-03-09 by 0xc0170