ARMmbed/mbed-os #3793: Exported project does not link in Embitz

vargham, 2017-02-17:
There is an unwanted option in the exported project: -c in linker options. It must be removed every time when a project exported to Embitz.

> -c Compile or assemble the source files, but do not link. The linking stage simply is not done.
[GCC doc: Options Controlling the Kind of Output](https://gcc.gnu.org/onlinedocs/gcc/Overall-Options.html)

### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
Maybe all (Tried with STM32)

**Toolchain:**
GCC_ARM

**Expected behavior**
Complete build process should be done.

**Actual behavior**
Build process stops at linking.

**Steps to reproduce**
Create a project in the online IDE and export to Embitz.

0xc0170, 2017-02-20:
cc @theotherjimmy

theotherjimmy, 2017-02-21:
I could not reproduce this error. 
The relevent section of the `eix` file:
```xml
            <LD>
                <Switch name="-Wl,--gc-sections"/>
                <Switch name="-Wl,--wrap,main"/>
                <Switch name="-Wl,--wrap,_malloc_r"/>
                <Switch name="-Wl,--wrap,_free_r"/>
                <Switch name="-Wl,--wrap,_realloc_r"/>
                <Switch name="-Wl,--wrap,_calloc_r"/>
                <Switch name="-Wl,--wrap,exit"/>
                <Switch name="-Wl,--wrap,atexit"/>
                <Switch name="-mcpu=cortex-m4"/>
                <Switch name="-mthumb"/>
                <Switch name="-mfpu=fpv4-sp-d16"/>
                <Switch name="-mfloat-abi=softfp"/>
                
            </LD>
```

no `-c`

vargham, 2017-02-22:
I tried again. Target: LPC1768, mbed-dev. Exported eix file before opened with EmBlocks:
```
            <LD>
                <Switch name="-c"/>
                <Switch name="-Wall"/>
                <Switch name="-Wextra"/>
                <Switch name="-Wno-unused-parameter"/>
                <Switch name="-Wno-missing-field-initializers"/>
                <Switch name="-fmessage-length=0"/>
                <Switch name="-fno-exceptions"/>
                <Switch name="-fno-builtin"/>
                <Switch name="-ffunction-sections"/>
                <Switch name="-fdata-sections"/>
                <Switch name="-funsigned-char"/>
                <Switch name="-MMD"/>
                <Switch name="-fno-delete-null-pointer-checks"/>
                <Switch name="-fomit-frame-pointer"/>
                <Switch name="-Os"/>
                <Switch name="-mcpu=cortex-m3"/>
                <Switch name="-mthumb"/>
                <Switch name="-Wl,--gc-sections"/>
                <Switch name="-Wl,--wrap,main"/>
                <Switch name="-Wl,--wrap,_malloc_r"/>
                <Switch name="-Wl,--wrap,_free_r"/>
                <Switch name="-Wl,--wrap,_realloc_r"/>
                <Switch name="-Wl,--wrap,_calloc_r"/>
                <Switch name="-Wl,--wrap,exit"/>
                <Switch name="-Wl,--wrap,atexit"/>
                <Switch name="-mcpu=cortex-m3"/>
                <Switch name="-mthumb"/>
                
            </LD>
```
Target Nucleo F103RB, mbed-lib. Exported eix file before opened with EmBlocks:
```
            <LD>
                <Switch name="-c"/>
                <Switch name="-Wall"/>
                <Switch name="-Wextra"/>
                <Switch name="-Wno-unused-parameter"/>
                <Switch name="-Wno-missing-field-initializers"/>
                <Switch name="-fmessage-length=0"/>
                <Switch name="-fno-exceptions"/>
                <Switch name="-fno-builtin"/>
                <Switch name="-ffunction-sections"/>
                <Switch name="-fdata-sections"/>
                <Switch name="-funsigned-char"/>
                <Switch name="-MMD"/>
                <Switch name="-fno-delete-null-pointer-checks"/>
                <Switch name="-fomit-frame-pointer"/>
                <Switch name="-Os"/>
                <Switch name="-mcpu=cortex-m3"/>
                <Switch name="-mthumb"/>
                <Switch name="-Wl,--gc-sections"/>
                <Switch name="-Wl,--wrap,main"/>
                <Switch name="-Wl,--wrap,_malloc_r"/>
                <Switch name="-Wl,--wrap,_free_r"/>
                <Switch name="-Wl,--wrap,_realloc_r"/>
                <Switch name="-Wl,--wrap,_calloc_r"/>
                <Switch name="-Wl,--wrap,exit"/>
                <Switch name="-Wl,--wrap,atexit"/>
                <Switch name="-mcpu=cortex-m3"/>
                <Switch name="-mthumb"/>
                
            </LD>
```
[image]

vargham, 2017-02-23:
I use the online IDE to export project.
Should I provide more information?

theotherjimmy, 2017-02-23:
was fixed by https://github.com/ARMmbed/mbed-os/pull/3521

theotherjimmy, 2017-02-23:
That fix went into 5.3.3. The online tools are scheduled for an update soon, and that update will use version 5.3.5 or later, brining in this fix.

vargham, 2017-02-23:
OK, Thank you.

closed 2017-06-18 by vargham