ARMmbed/mbed-os #3804: Heap stats not reported correctly with IAR in NUMAKER_PFM_NUC472

andresag01, 2017-02-20:
While trying to profile the heap usage by one of the sample applications I used the heap_stats feature from mbed OS as described in [this website](https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/advanced/runtime_stats/). Specifically, I ran the following code:

```C++
#include "mbed.h"
#include "mbed_stats.h"
#include "cmsis_os.h"

int main(void)
{
	mbed_stats_heap_t heap_stats;

	printf("Starting head stats example\r\n");
	printf("Starting stack stats example\r\n");

	osThreadId main_id = osThreadGetId();

	osEvent info;
	info = _osThreadGetInfo(main_id, osThreadInfoStackSize);
	if (info.status != osOK) {
		printf("Could not get stack size\r\n");
	}
	uint32_t stack_size = (uint32_t)info.value.v;
	info = _osThreadGetInfo(main_id, osThreadInfoStackMax);
	if(info.status != osOK) {
		printf("Could not get max stack\r\n");
	}
	uint32_t max_stack = (uint32_t)info.value.v;
	printf("Stack used %li of %li bytes\r\n", max_stack, stack_size);

	void *allocation = malloc(1000);
	printf("Freeing 1000 bytes\r\n");

	mbed_stats_heap_get(&heap_stats);
	printf("Current heap: %lu\r\n", heap_stats.current_size);
	printf("Max heap size: %lu\r\n", heap_stats.max_size);

	free(allocation);

	mbed_stats_heap_get(&heap_stats);
	printf("Current heap after: %lu\r\n", heap_stats.current_size);
	printf("Max heap size after: %lu\r\n", heap_stats.max_size);

	return 0;
}
```
Then I compiled using the command:
```
mbed compile -t IAR -m NUMAKER_PFM_NUC472 -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -c
```
And observed the following output in the serial terminal:
```
Starting head stats example
Starting stack stats example
Stack used 392 of 4096 bytes
Freeing 1000 bytes
Current heap: 0
Max heap size: 0
Current heap after: 0
Max heap size after: 0
```
However, this output is not accurate as my program consumes at least 1KB of heap memory with that call to `malloc()`. Note that the statistics reported seem correct when compiling with ARM and GCC_ARM.

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
IAR

**Toolchain version:**
IAR ANSI C/C++ Compiler V7.70.1.11437/W32 for ARM

**mbed-cli version:** 
1.0.0

**meed-os sha:**
#bcf7085d85b2811b5d68bdda192c754eadfb8f88

**Expected behavior**
Heap stats should be reported correctly.

**Actual behavior**
Heap usage is always reported as 0 bytes.

**Steps to reproduce**
Compile (using IAR) and run the program pasted at the beginning of this report on the NUMAKER_PFM_NUC472.

0xc0170, 2017-02-20:
cc @c1728p9

c1728p9, 2017-02-20:
Heap stats never worked for IAR, since there isn't a way to hook their functions

deepikabhavnani, 2017-09-21:
Resolved in PR #4965

closed 2018-06-02 by deepikabhavnani