ARMmbed/mbed-os #3805: BR & mbed-client example not working anymore together

betzw, 2017-02-20:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F429ZI

**Toolchain:**
GCC_ARM

**meed-os sha:**
Discovered in: `6ca86a7a24b5df4078077061ff59bb080ce6b527`
Responsible sha most likely: `667006e1ba661cb64d58419c67ba623d86c4dbec`

**Actual behavior**
"No beacons" error on mbed-client side

Seems that with Nanostack v6.0.1 something has changed in a way that border-router doesn't work anymore together with mbed-client on platform combination (for both applications): NUCLEO_F429ZI + Spirit1 RF. mbed-client report "No Beacons".

cc @SeppoTakalo

SeppoTakalo, 2017-02-24:
I cannot replicate this.

I tried with Nanostack-border router using mbed-OS master, and mbed-client using one commit behind mbed-OS master.

```
[DBG ][mlme]: MAC: Start network 23 channel 692 panid
[DBG ][6Bo ]: Mac Ready
[DBG ][6Bo ]: SET Security Mode
[DBG ][mlme]: 01
[DBG ][mlme]: Free PAN result
[DBG ][mlme]: Free Response
[DBG ][6Bo ]: Accept & Request
[DBG ][MACh]: Register Device
[DBG ][mMIB]: Over write 0, mac16 ffff mac64: ff:ff:ff:ff:ff:ff:ff:ff, 4294967295
[DBG ][mMIB]: Set 0, mac16 face mac64: fc:c2:3d:00:00:04:a2:93, 109
[DBG ][6Bo ]: RS*
[DBG ][mMCp]: Set new device user 0 for key 00:a1:a2:a3:a4:a5:a6:a7:a8:a9:aa:ab:ac:ad:ae:af
[DBG ][loND]: Create ND version
[DBG ][rout]: Added route:
[DBG ][rout]:  2001:XXX:XXX:6f5:fec2:3d00:4:7a5b/128 if:1 src:'Loopback' id:0 lifetime:infinite
[DBG ][rout]:      On-link (met 128)
[INFO][addr]: Address added to IF 1: 2001:470:28:6f5:fec2:3d00:4:7a5b
```

Can you try your setup with full logging enabled and `FEATURE_NANOSTACK_FULL` enabled as it outputs more logs.

betzw, 2017-02-27:
Today I have tried with `nanostack-border-router-private` and `mbed-os` (branch `master`, sha1: `3a27568a505bbc0bb8eeb973109d3d39ce823d1c`. I have tracing enabled
```"debug-trace": "true",``` & ```"mbed-trace.enable": 1,```
and```"target.features_add": ["NANOSTACK", "NANOSTACK_FULL", "COMMON_PAL"],```

The whole resulting trace comign out from the border router is:
```
[DBG ][evlp]: event_loop_thread
...
[INFO][brro]: PANID: 691
[INFO][brro]: NET_IPV6_BOOTSTRAP_AUTONOMOUS
[DBG ][brro]: Using PSK security mode, key ID = 1
[DBG ][core]: Allocate Root Tasklet
[DBG ][core]: NS Root task Init
[90m[DBG ][sck ]: Socket Task
[DBG ][pana]: Ini[DBG ][pana]: Init
sklet Generated
t
[DBG ][sck ]: Socket id 0 allocated
[DBG ][pana]: Pana Tasklet Init
```

Note, I am using the nanostack, mbed-client, etc. binaries coming with mbed-os!
P.S.: Platform combination is `NUCLEO_F429ZI` + `X-NUCLEO-IDS01A4` (a.k.a. Spirit1).

closed 2017-03-01 by betzw