ARMmbed/mbed-os #3807: cmsis5: Export to uvision5 doesn't compile for ST

bulislaw, 2017-02-20:
### Description
- Type: Bug 
- Priority: Blocker
- Branch: https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5
- Related: #3683 

---------------------------------------------------------------
## Bug

**Target**
Nucleo F401RE

**Toolchain:**
ARM

**Toolchain version:**
uVision 5.2
ARMC 5.06

**Expected behavior**

It compiles for both ARMC and uVision

**Actual behavior**

It works fine for ARMC:
```
 mbed compile -t arm -m nucleo_f401re -c
[...]
Link: mbed-os
Elf2Bin: mbed-os
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Misc      | 36516 |  1725 | 5808 |
| Subtotals | 36516 |  1725 | 5808 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 7533 bytes
Total RAM memory (data + bss + heap + stack): 7533 bytes
Total Flash memory (text + data + misc): 38241 bytes

Image: .\BUILD\nucleo_f401re\arm\mbed-os.bin
```

but for uVision:
```
 mbed export -m nucleo_f401re -i uvision5
```
I'm getting errors:
```
assembling irq_cm4f.s...
rtos\rtx2\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM\irq_cm4f.s(81): error: A1854E: Unknown opcode 'VSTMDBEQ', maybe wrong target CPU?
rtos\rtx2\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM\irq_cm4f.s(97): error: A1854E: Unknown opcode 'VLDMIAEQ', maybe wrong target CPU?
```

F401 has FPU so that should work, not to mention that it works fine for CLI ARMC build. It's a blocker for cmsis update, as ST guys can't really debug their boards with the feature branch.

@theotherjimmy

0xc0170, 2017-02-20:
I can reproduce the problem. I noticed that FPU settings are not active in the project file. Once I select again the same MCU ,they appear and the code compiles.

>  It's a blocker for cmsis update, as ST guys can't really debug their boards with the feature branch.

They could for now use  the trick I did (select MCU target again to unlock FPU options). We will fix this

bulislaw, 2017-02-20:
@bcostm You can either use workaround suggested by @0xc0170 or one proposed by me in #3683

0xc0170, 2017-02-20:
Seems like ``<Cpu>IRAM(0x20000000,0x18000) IROM(0x08000000,0x80000) CPUTYPE("Cortex-M4") CLOCK(12000000) ELITTLE</Cpu>`` is incorrect for MCU with FPU. Once I add FPU2  (this is FPU type) the option appears.

Looking at uvision exporter, the line 200 might need some work:

```
ctx['cputype'] = ctx['device'].core.rstrip("FD")
```

This is set as cputype, should it include FPU settings as well?

@theotherjimmy this should help you

0xc0170, 2017-03-01:
@theotherjimmy Please look at this one, you can see above that it causing problems during OOB (any target with FPU cant be compiled in the uvision IDE)

theotherjimmy, 2017-03-01:
Thanks for organizing this for me @0xc0170 I'll tag these issues in the PR description.

PR #3861 (2017-03-03): Add the FPU field to the <Cpu> tag in Keil
Recently Keil released version 5.23 of their IDE. This version requires
the FPU to be set as part of the <Cpu> tag in the .uvprojx (XML project
file). This patch adds the appropriate FPU settings based on the
trailing F (FPU enabled) or FD (Double precesion FPU enabled) string of
the core.

## Related issues:
  - Resolves #3807 
  - Resolves https://github.com/ARMmbed/mbed-os-example-wifi/issues/19
  - Resolves https://github.com/ARMmbed/mbed-os-example-ble/issues/56
  - Resolves https://github.com/ARMmbed/mbed-os-example-sockets/issues/16
  - Resolve [... cut]
closed 2017-03-03 by 0xc0170