ARMmbed/mbed-os #3808: setting and overriding compilation flags according to target

RonEld, 2017-02-20:
### Description
- Type:  Enhancement 
- Priority: Major

I need an ability to override some compilation flags, according to target. I am trying to port mbed TLS over HW cryptographic accelerator, which requires -mfloat=hard as a compilation flag. The only place I saw that this is handle is in tools/toolchains/gcc.py:        
 if target.core == "Cortex-M4F":
   self.cpu.append("-mfpu=fpv4-sp-d16")
   self.cpu.append("-mfloat-abi=softfp")

and it doesn't seem to be able to be overridden.
for the moment, I modified this value, so I won't be blocked, but obviously this is not the way to handle this requirement.

0xc0170, 2017-02-20:
Correct that is hardcoded in the gcc.

cc @theotherjimmy

theotherjimmy, 2017-02-21:
And there is no right way to handle this ATM. You are the first person I'm aware of that has requested supporte for `-mfloat-abi=hard`.

geky, 2017-02-21:
The name `-mfloat-abi=softfp` is awfully terrible. It enables hardware floating point operations with the _calling convention_ of software floating point (necessary for binary compatiblity with external libraries).

Is there a non-obvious reason for `-mfloat-abi=hard`? Are you trying to save stack space for the floating point arguments?

RonEld, 2017-02-22:
@geky The short answer is that without setting `-mfloat-abi=hard` I get link error:
`
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ./mbed-os/features/mbedtls/targets/TARGET_CRYPTOCELL310/TARGET_NRF5_SDK13/TOOLCHAIN_GCC_ARM/libcc310_gcc_0.9.0.a(ssi_pal_mutex.c.obj) uses VFP register arguments, ./BUILD/NRF52840_DK/GCC_ARM/authcrypt.elf does not`
This was done with the precompiled cryptocell library delivered by partner.

I have tried compiling the cryptocell library with `softfp` and I didn't get any link error, but the device failed to start, and it was stuck at the startup script.

My idea is not only for mabi option, but an enhancement to support overriding compilation flags according to target, and not rely on profile alone.
At the moment I am a bit blocked by the mabi option.

RonEld, 2017-02-22:
Hi, after I recompiled the cryptocell library, and fixed all configurations there, I managed to run on the board with `-mabi=softfp`, but this is only because I have the cryptocell source code. @screamerbg we should update partner to modify the precompiled library without -mabi=hard, for mbed-os

I still think we should have a support to allow overriding compilation flags, in targets.json , on cases where there are precopmiled binaries for example

geky, 2017-02-22:
Ah interesting, abi compatibility with libraries is a valid reason to change this flag.

There are some lingering concerns with binaries already in mbed OS. Though any issues there will at least be revealed at compile time.

RonEld, 2017-02-23:
I agree with the concern, because it is a vast system, with many components.  
At the moment we can keep this issue at low priority, as we can ask the current partner to supply the precompiled library without the `-mabi=hard` flag.   
Perhaps we should mark this flag as a requirement for all components, and re-discuss this issue if there will be a component that will not work with `softfp` abi flag.

0xc0170, 2017-02-23:
>  Perhaps we should mark this flag as a requirement for all components, and re-discuss this issue if there will be a component that will not work with softfp abi flag.

+1

cc @sg-

theotherjimmy, 2017-04-25:
@RonEld Is the status of this issue "We need to document that we use `-mfloat-abi=softfp`"?

RonEld, 2017-04-26:
@theotherjimmy I think that at the moment, this is the status, unless there will be reasonable objections from future partners. During latest SIP workshop, our partner didn't want to remove the `-mabi=hard` compilation flag, and they said they need further investigation on this, from their side

theotherjimmy, 2017-04-26:
@AnotherButler Where could we document this requirement?

AnotherButler, 2017-04-26:
Probably the porting guide section, but I'm open to other suggestions

theotherjimmy, 2017-04-26:
Thanks @RonEld I'll mention you on the PR I make to the porting guide section and mention that PR here.

theotherjimmy, 2017-04-26:
@RonEld can we close this issue? I have the PR up to document these requirements.

RonEld, 2017-04-27:
@theotherjimmy I saw the other PR, and for now it seems that this issue can be closed. However, I think that we should have this issue "on hold", so it won't be lost, because I know of partners that have their code in `mabi=hard` and they are investigating if it's acceptable on their side to change to `softfp`.
As mentioned earlier
> Perhaps we should mark this flag as a requirement for all components, and re-discuss this issue if there will be a component that will not work with `softfp` abi flag.

Is it possible to add some sort of label to be easier tracked in the future?

theotherjimmy, 2017-04-27:
Closing the issue does not remove it from github. Sure. We could, but I don't think we have such a label right now.

RonEld, 2017-05-03:
@theotherjimmy I think we can close this issue, as long as we have this requirement tracked somewhere, in low priority . (perhaps in JIRA?)

theotherjimmy, 2017-05-03:
Yeah, we do. I'll close it then.

closed 2017-05-03 by theotherjimmy