ARMmbed/mbed-os #3809: NUCLEO_L053R8 not showing up in mbed CLI as mbed OS 2 supported

MarceloSalazar, 2017-02-20:
### Description
- Type: Bug
- Priority: Blocker

## Bug

**Target**
NUCLEO_L053R8

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.0.0

**meed-os sha:**
https://mbed.org/users/mbed_official/code/mbed/builds/ef9c61f8c49f   (Release 136)

**Expected behavior**
Device should show up as mbed OS 2 supported when using mbed CLI and compiling

**Actual behavior**
Device not included in release

**Steps to reproduce**

```
> mbed new mbed-classic-program --mbedlib
> cd mbed-classic-program
> mbed compile -S

```

However, I can see the platform NUCLEO_L053R8 has been added to mbed OS 5.3.5:
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.3.5/targets/targets.json#L1015

@theotherjimmy @adustm

jeromecoutant, 2017-02-21:
Hi

My test:
- the default "mbed compile -S" command shows all the MBED OS 5 platforms

- If I patch  tools\build_api.py file:

def mcu_toolchain_matrix(verbose_html=False, platform_filter=None,
-                          release_version='5'):
+                          release_version='2'):

"mbed compile -S" command shows now all the MBED OS 2 platforms (including NUCLEO_L053R8)

0xc0170, 2017-02-21:
As @jeromecoutant pinpointed, I don't think ``mbed cli`` should show anything mbed 2 related. Therefore this platform is not supported (it's only for mbed 2).

MarceloSalazar, 2017-02-21:
Indeed, mbed CLI can import and build both mbed OS 5 and 2 versions. The device should show up as supported in mbed OS 2. There is already a column for mbed OS 2:

```
>mbed compile -S
+----------------------+-----------+-----------+-----------+-----------+-----------+
| Target               | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |
+----------------------+-----------+-----------+-----------+-----------+-----------+
| xxx             | Supported | Supported | xx | xx | xx |
```

sg-, 2017-04-19:
@bridadan There was a reason this wasnt the case by default but I dont recall. Was it a todo with enhanced filtering when -S is used?

bridadan, 2017-04-19:
@sg- This is because by default mbed CLI only shows mbed OS 5 supported platforms. I'm actually not sure if we have a publicly exposed way of showing mbed 2 targets (via mbed CLI). The capability exists in the code base though. This is the function: https://github.com/ARMmbed/mbed-os/blob/master/tools/build_api.py#L1103

theotherjimmy, 2018-06-25:
@studavekar Could we make the CI smart enough to check for mbed 5 support? Then I can resolve this issue.

studavekar, 2018-06-25:
> @studavekar Could we make the CI smart enough to check for mbed 5 support? Then I can resolve this issue.

we already pick devices which are in mbed 5 supported column.

theotherjimmy, 2018-06-25:
Sorry for implying othewise. We can change the output to include mbed2 then?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-99*

deepikabhavnani, 2018-12-28:
As per Jira this issue was resolved by https://github.com/ARMmbed/mbed-os/pull/7396

closed 2018-12-28 by deepikabhavnani