ARMmbed/mbed-os #3811: mbed test suite for mbed Enabled for mbed OS 2 ("Classic")

iotvietmember, 2017-02-20:
### Description
- Type:  Question
- Related issue: `#mbed test suite` `#mbed Enabled`

----------------------------------------------------------
## Question
I want to apply for my board (Nordic nRF51822) to be accredited as an official ARM mbed Enabled product (mbed Enabled for mbed OS 2)

Technical Requirements:
https://www.mbed.com/en/about-mbed/mbed-enabled/mbed-enabled-program-requirements/#Technical_Requirements

```
Require 3: Hardware platform testing
Passes all tests in the mbed test suite (http://github.com/mbedmicro/mbed)
```
But I could not found tutorial to do.
**How to run mbed test suite(mbed OS 2) for my board to apply mbed Enabled ?**

Thank you very much!

0xc0170, 2017-02-21:
cc @sg- @MarceloSalazar @BlackstoneEngineering

MarceloSalazar, 2017-02-21:
For this kind of enquires, please write to us: info@mbed.com

closed 2017-02-21 by MarceloSalazar