ARMmbed/mbed-os #3813: Online compiler does not add -D__MICROLIB --library_type=microlib option for "small" build target

toyowata, 2017-02-21:
### Description
- Type: Bug
- Related issue: https://developer.mbed.org/questions/77071/Offline-Compliling/
- Priority: Blocker

---------------------------------------------------------------
## Bug

Recently, online compiler doesn't add ```-D__MICROLIB --library_type=microlib``` option when small target (-t uARM) is compiled.

[Offline build - build.py -m LPC812 -t uARM]
```
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -D__MBED_CMSIS_RTOS_CM -DTARGET_LPC81X -DTOOLCHAIN_object -D__CMSIS_RTOS -DTARGET_CORTEX_M -DTARGET_M0P -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DDEVICE_I2C=1 -D__CORTEX_M0PLUS -DTOOLCHAIN_ARM_MICRO -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DARM_MATH_CM0PLUS -DDEVICE_SLEEP=1 -DTARGET_LPC812 -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1487662711.54 -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0
Compile [100.0%]: main.cpp
[DEBUG] Compile: C:\Keil_v5\ARM\ARMCC\bin\armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 -D__MICROLIB --library_type=microlib -DMBED_RTOS_SINGLE_THREAD --cpu=Cortex-M0 --md --no_depend_system_headers --c99 -D__ASSERT_MSG --cpp --no_rtti --no_vla -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -D__MBED_CMSIS_RTOS_CM -DTARGET_LPC81X -DTOOLCHAIN_object -D__CMSIS_RTOS -DTARGET_CORTEX_M -DTARGET_M0P -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DDEVICE_I2C=1 -D__CORTEX_M0PLUS -DTOOLCHAIN_ARM_MICRO -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DARM_MATH_CM0PLUS -DDEVICE_SLEEP=1 -DTARGET_LPC812 -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1487662711.54 -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0 --via .\BUILD\LPC812\uarm\.includes_b2c5a7b1610a4f675582b7415fb931f8.txt --preinclude=.\BUILD\LPC812\uarm\mbed_config.h --depend .\BUILD\LPC812\uarm\.\main.d -o .\BUILD\LPC812\uarm\.\main.o .\main.cpp
```

[Online build - NXP LPC800-MAX target (LPC812)]
```
Info: Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -D__MBED_CMSIS_RTOS_CM -DTARGET_LPC81X -DTOOLCHAIN_object -D__CMSIS_RTOS -DMBED_USERNAME=MACRUM -DTARGET_CORTEX_M -DTARGET_M0P -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -D__CORTEX_M0PLUS -DTOOLCHAIN_ARM_MICRO -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DARM_MATH_CM0PLUS -DDEVICE_SLEEP=1 -DTARGET_LPC812 -DMBED_BUILD_TIMESTAMP=1487662778.73 -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_PWMOUT=1 -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0
Info: Compile: armcc -c --gnu -O3 -Otime --split_sections --apcs=interwork --cpu=Cortex-M0 --md --no_depend_system_headers --c99 -D__ASSERT_MSG --cpp --no_rtti -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -D__MBED_CMSIS_RTOS_CM -DTARGET_LPC81X -DTOOLCHAIN_object -D__CMSIS_RTOS -DMBED_USERNAME=MACRUM -DTARGET_CORTEX_M -DTARGET_M0P -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -D__CORTEX_M0PLUS -DTOOLCHAIN_ARM_MICRO -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DARM_MATH_CM0PLUS -DDEVICE_SLEEP=1 -DTARGET_LPC812 -DMBED_BUILD_TIMESTAMP=1487662778.73 -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_PWMOUT=1 -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0 -I/extras -I/src -I/source/mbed-dev -I/source/mbed-dev/cmsis -I/source/mbed-dev/drivers -I/source/mbed-dev/hal -I/source/mbed-dev/hal/storage_abstraction -I/source/mbed-dev/platform -I/source/mbed-dev/targets -I/source/mbed-dev/targets/TARGET_NXP -I/source/mbed-dev/targets/TARGET_NXP/TARGET_LPC81X -I/source/mbed-dev/targets/TARGET_NXP/TARGET_LPC81X/TARGET_LPC812 -I/source/mbed-dev/targets/TARGET_NXP/TARGET_LPC81X/TARGET_LPC812/device -I/source/mbed-dev/targets/TARGET_NXP/TARGET_LPC81X/TARGET_LPC812/device/TOOLCHAIN_ARM_MICRO -I/source/mbed-dev/targets/TARGET_NXP/TARGET_LPC81X/device -I/source/mbed-dev/targets/TARGET_NXP/TARGET_LPC81X/device/TOOLCHAIN_ARM_MICRO --preinclude=/build/mbed_config.h --depend /build/./main.LPC812.d -o /build/./main.LPC812.o /source/main.cpp in "main.cpp", Line: 0, Col: 1
```

The result of this makes large size binary image and it is critical for small ROM size target.  I got out of memory error by few of my programs by mbed-dev.

This doesn't depend on version of mbed or mbed-dev library, since this maybe caused by online compiler build script?

**Target**
Any ARM small build target (uARM)

**Toolchain:**
uARM

**Expected behavior**

Build by ARMCC micro library.

**Actual behavior**

Build by ARMCC standard library.

**Steps to reproduce**

* Create new "Blinky LED Hello World" program target to NXP LPC800-MAX (or any small build target) 
* Check "Detail" box in the status pane
* Compile the program and see build information

0xc0170, 2017-02-21:
cc @theotherjimmy

toyowata, 2017-03-03:
Related bug report : https://developer.mbed.org/questions/77234/mbed-LPC1114FN28mbed-dev/

0xc0170, 2017-03-15:
@theotherjimmy Any update?

theotherjimmy, 2017-03-15:
It's actually a pretty simple problem: the online compiler just assumes that all targets are compiled with ARM and none with uARM.

theotherjimmy, 2017-03-15:
I posted a fix. If you have access, you should see it.

toyowata, 2017-03-16:
@theotherjimmy Thank you very much for the fix.  The online compiler does not seem to update yet.  Do you know when it will be deployed to the online service?

theotherjimmy, 2017-03-20:
@toyowata I do not. I will inform you when it's up.

toyowata, 2017-04-05:
Confirmed the fix. Thanks a lot.

closed 2017-04-05 by toyowata