ARMmbed/mbed-os #3815: mbed-dev generates bigger binary than mbed

janjongboom, 2017-02-21:
### Description
- Type: Question

## Question

Message from the [Q&A section](https://developer.mbed.org/questions/77071/Offline-Compliling/#answer12120?compage=1#c26420). Apparently compiling against mbed-dev generates a binary 4K bigger than compiling against mbed. Do we use less optimizations in the online compiler for user apps than we do for pre-compiling mbed?

@sg-

0xc0170, 2017-02-21:
I don't recall any special settings, both of them use the same profile (=same settings). However linker might do some different magic library vs object files (to resolve weak/strong symbols, etc). 
I cant spot there a reference for 4K size difference. How was this size measured?

cc @theotherjimmy

toyowata, 2017-02-21:
Probably it related this: https://github.com/ARMmbed/mbed-os/issues/3813

The mbed library has pre-build object files for small build (uARM), but mbed-dev source file are compiled by standard build (ARM).  That's why you will get bigger bin image by mbed-dev compare with (pre-build) mbed library.

0xc0170, 2017-02-21:
@toyowata Thanks, that would explain it if a user uses target that defaults to use uARM.

theotherjimmy, 2017-04-26:
@janjongboom Can we close this issue? I fixed the usage of uARM on the online compiler a while back.

closed 2017-05-01 by janjongboom