ARMmbed/mbed-os #3819: EFM32_STK3200 does not compile with mbed-cli

pfried, 2017-02-21:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
EFM32_STK3200

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc version 5.3.0 (GCC)

**mbed-cli version:** 
1.0.0

**meed-os sha:**
269f58d Merge pull request #3660 from ARMmbed/release-candidate

**Expected behavior**
It should compile

**Actual behavior**
Gives an error, see output below

**Steps to reproduce**
run `mbed-compile` with a new project based on the EFM32ZG_STK3200 target. main.cpp is the blinky default program

----------------------------------------------------------------
## Suspicion

The RTC_Conf_CM.c includes the mbed_rtx.h file provided by the platform (at targets/TARGET_Silicon_Labs/mbed_rtx.h). This file includes configuration like:

```
#elif defined(TARGET_EFM32HG_STK3400)

#ifndef INITIAL_SP
#define INITIAL_SP              (0x20002000UL)
#endif

#ifndef OS_TASKCNT
#define OS_TASKCNT              6
#endif
#ifndef OS_MAINSTKSIZE
#define OS_MAINSTKSIZE          112
#endif
```
for all target STKs but the STK3200 as there is no definition for EFM32ZG_STK3200 or a parent. I suspect that it is simply a forgotten definition.

-----------------------------------------------------------------

Output from `mbed compile`

```
Compile [ 66.2%]: RTX_Conf_CM.c
[Error] RTX_Conf_CM.c@74,3: #error "no target defined"
[Error] RTX_CM_lib.h@438,6: #error "no target defined"
[Error] RTX_CM_lib.h@452,55: 'INITIAL_SP' undeclared (first use in this function)
[Error] RTX_CM_lib.h@458,39: 'OS_MAINSTKSIZE' undeclared (first use in this function)
[ERROR] .\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_Conf_CM.c:74:3: error: #error "no target defined"
  #error "no target defined"
   ^
In file included from .\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_Conf_CM.c:277:0:
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h:438:6: error: #error "no target defined"
     #error "no target defined"
      ^
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h: In function 'set_stack_heap':
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h:452:55: error: 'INITIAL_SP' undeclared (first use in this function)
         #define HEAP_SIZE       ((uint32_t)((uint32_t)INITIAL_SP - (uint32_t)HEAP_START))
                                                       ^
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h:473:26: note: in expansion of macro 'HEAP_SIZE'
     uint32_t free_size = HEAP_SIZE;
                          ^
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h:452:55: note: each undeclared identifier is reported only once for each function it appears in
         #define HEAP_SIZE       ((uint32_t)((uint32_t)INITIAL_SP - (uint32_t)HEAP_START))
                                                       ^
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h:473:26: note: in expansion of macro 'HEAP_SIZE'
     uint32_t free_size = HEAP_SIZE;
                          ^
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h:458:39: error: 'OS_MAINSTKSIZE' undeclared (first use in this function)
     #define ISR_STACK_SIZE ((uint32_t)OS_MAINSTKSIZE * 4)
                                       ^
.\mbed-os\rtos\rtx\TARGET_CORTEX_M\RTX_CM_lib.h:481:27: note: in expansion of macro 'ISR_STACK_SIZE'
     mbed_stack_isr_size = ISR_STACK_SIZE;
                           ^

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\mbed\blinky-zero-gecko\mbed-os\tools\make.py -t GCC_ARM -m EFM32ZG_STK3200 --source . --build .\BUILD\EFM32ZG_STK3200\GCC_ARM" in "C:\mbed\blinky-zero-gec

```

0xc0170, 2017-02-22:
cc @stevew817

stevew817, 2017-02-22:
Running RTOS on Zero Gecko doesn't make a whole lot of sense due to the very constrained RAM... Which is why, in targets.json, it is only set to release on mbed 2(classic), not 5(OS). I can add RTOS support for Zero, but am asking how useful it would be.

pfried, 2017-02-22:
Even if it does not: We do have several development kits and i wanted to setup a blinky example on them, so i followed a simple workflow:

1. `mbed new myproject`
2. `mbed target EFM32ZG_STK3200 && mbed toolchain GCC_ARM`
3. Create a simple blinky `main.cpp`
4. `mbed compile`

You are right in the `targets.json` it is only set as release version 2, but people not using the online compiler will not really see this, as on the board page there is no real hint about that. It says `ARM mbed enabled` and that`s it. For me mbed os is not only about the RTOS, but also about the tooling around it.

Either it gives a warning or error when trying to compile it for mbed os 5 (low ram or so) or it tweaks the configuration to a low profile or disable the RTOS at all (if that is possible, i am not yet familiar with the framework).

Everything is better than cryptic error messages from the compiler

stevew817, 2017-02-22:
@0xc0170 Isn't there a CLI option you can set to compile with the v2 library instead of v5? This really should be caught in the CLI, since I assume other platforms that are v2-only would have the same issue.

0xc0170, 2017-02-23:
> Either it gives a warning or error when trying to compile it for mbed os 5 (low ram or so) or it tweaks the configuration to a low profile or disable the RTOS at all (if that is possible, i am not yet familiar with the framework). 

> bedEverything is better than cryptic error messages from the compiler

cc @bridadan @theotherjimmy 

> @0xc0170 Isn't there a CLI option you can set to compile with the v2 library instead of v5? This really should be caught in the CLI, since I assume other platforms that are v2-only would have the same issue.

This one: https://github.com/ARMmbed/mbed-cli#creating-a-new-program-for-mbed-os-2

stevew817, 2017-02-23:
@pfried Can you compile using the --mbedlib flag @0xc0170 linked to? That should solve your problem for the moment. I agree there should be a better message, but that is something for the mbed team to solve in the CLI and/or during target discovery.

@0xc0170 This issue is not specific to Silicon Labs devices anymore. Do we still need that tag here?

pfried, 2017-03-09:
that it okay for me, we only used the board for evaluation, so it does not really matter

closed 2017-10-27 by ChiefBureaucraticOfficer