ARMmbed/mbed-os #3824: Generic CallChains

ptpaterson, 2017-02-22:
### Make CallChain class generic so that it can be used with different Callback templates.
- Type: Enhancement

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
CallChain classes are needed with function prototypes other than void(void).

**Suggested enhancement**
Expand definition of CallChain class similar to the Callback class with templates for
- R(A0, A1, A2, A3, A4)
- R(A0, A1, A2, A3)
- R(A0, A1)
- R(A0)
- R()

Initially possible to return the value of the last caller, but can build in aggregator class, similar to boost::signals2 to allow greater functionality with return values.

**Pros**
New functionality would be additional and should not interfere with existing application code.  

I have been using such an extension of the CallChain with several different callback function prototypes without editing the original non-template class code.

**Cons**
Convention with the templated versions would be slightly different:

- Each template needs its own version of `pFunctionPointer_t` and `CallChainLink`, therefore they need to be encapsulated into the template class.  I think that it should be okay to keep the global `pFunctionPointer_t`, and then each templated CallChain class would have its own typedef'd function pointers, (E.g. `CallChain<int(int, int)>::pFunctionPointer_t`) and `CallChainLink` classes.

Template specialization can be a pain to maintain?

**Other Notes**
The Callback class is fully defined in its header.  I have implemented the entire CallChain template class in the header as well, but I do not know how others feel about this?  I think that the worst thing is that the CallChainLink class is exposed, 

I would like feedback from others before submitting a pull request.

geky, 2017-02-22:
A patch to update CallChain would be very cool 👍 The current class is mostly for internal use so it hasn't been a priority to update.

Unfortunately, there are concerns related to backwards compatiblity. Changing CallChain to a template requires a rename and the old class must be marked as deprecated for a release. (This was the reason we renamed FunctionPointer -> Callback).

---

Random thoughts:
> The Callback class is fully defined in its header. I have implemented the entire CallChain template class in the header as well

This is best for a templated class. Templates and separate compilation units don't work well together.

> I think that the worst thing is that the CallChainLink class is exposed

C++ has decent namespacing, you could move CallChainLink into the class definition, turning it into CallChain::CallChainLink (or CallChain::Link if you dislike redundancy).

> pFunctionPointer_t

Do we need to keep this around? Does anyone think `pFunctionPointer_t` is more readable than `Callback<void()>`? 

> Callback(CallChain &)

Couldn't help but notice CallChain is _conveniently_ one word of storage. The Callback class is designed to automatically wrap callable (`operator()` defined) classes if they fit in a word. In theory you could pass CallChains by value into attach functions, but this is currently disallowed (correctly so) by the privatized copy-constructor. Perhaps we should slap some reference counting on top and let users treat callchains like first class functions?

ptpaterson, 2017-02-22:
> `pFunctionPointer_t`

I think I would prefer to see the callback typedef'd and then explicitly return a pointer when necessary.  This would make functions to register callbacks simpler I think.

```c++
// CallChain.h
template <typename R, typename A0>
class CallChain<R(A0)> {
public:
    typedef Callback<R(A0)> Callback_t;
    ...
    Callback_t * add(Callback<R(A0)> func) { ... }
    ...
}

// MyClass.h with better typedefs
class MyClass {
public:
    typedef CallChain<int(int)> MyCallChain;
    MyCallChain::Callback_t * RegisterCallback(MyCallChain::Callback_t cb) {
        return chain_.add(cb);
    }
private:
    MyCallChain chain_;
}
```

Right now, trying not to change things to drastically I'm left with something more like this:

```c++
// MyClass.h using pFunctionPointer_t
class MyClass {
public:
    typedef Callback<int(int)> MyCallback;
    typedef CallChain<int(int)> MyCallChain;
    typedef MyCallChain::pFunctionPointer_t MyCallChainCallbackPtr;
    MyCallChainCallbackPtr RegisterCallback(MyCallback cb) {
        return chain_.add(cb);
    }
private:
    MyCallChain chain_;
}
```

which I find is confusing things, because I want to show that the return value is the same as what we are adding.

**Other effects**
I can't really see what the implications of changing this on other internal dependencies would be, though.

ptpaterson, 2017-02-22:
And if we completely rewrite it, I think that priorities would be a nice addition.  

At this very moment, I am comfortable with my current application not having callbacks sorted, but I imagine some others might.  

I don't think that it should be to hard to do an 'insert' into the list.

ptpaterson, 2017-02-22:
*REALLY*, I would like to have an aggregator to capture the return values on a per-application-defined basis.

boost::signals2 is set up such that the aggregator function can return in the middle of the callbacks.  *THIS* is the specific functionality that I need at the moment - That one callback could find an error and declare the end of the chain, or maybe a user application wants to add a callback to the front of the chain and halt the chain if it is satisfied with the result.

pan-, 2017-02-23:
@ptpaterson Thanks for your suggestion, it would be interesting to extend the Callchain class to non nullary function. 

> Expand definition of CallChain class similar to the Callback class with templates for

There's no need to define a template for every arity out there a single class template definition can be  handle handle all of them because the `CallChain` class is just delegating to the `Callback` class. The only templates "duplication" needed are template overloads of the functions realizing the call. I've made a quick prototype of it [here](https://gist.github.com/pan-/aa3519b0f6371f4f0c67058aa3fbe770).

> And if we completely rewrite it, I think that priorities would be a nice addition.
At this very moment, I am comfortable with my current application not having callbacks sorted, but I imagine some others might.

Do you want your callback sorted or ordered ? At the moment, they are already ordered by how the user inserts them. Generic (and extendable) sort of callbacks have the potential to increase the complexity of the interface and the code. 

> REALLY, I would like to have an aggregator to capture the return values on a per-application-defined basis.

I'm not sure it is necessary to add this responsibility to the class. The list of callbacks contains in the `CallChain` can be traversed from the outside so it is possible for user code to traverse the callback list, call each of them individually and do whatever is needed with the result.

ptpaterson, 2017-02-23:
**Template Specialization and Aggregator for `call`**
The benefit to building in an aggregator into the CallChain would be to have `call`/`()` operator return the same value type as the `Callback`s.  However, I realize both:

1. realize my use case is pretty specific, and that I can iterate over the callbacks with `size()` and `get()`, and
2. @pan, Your template version is indeed much more elegant, and versatile.  I agree it is the right direction.  It would then be difficult to extract the return value of `Callback::call`, which is as I say a big reason to otherwise build in an aggregator.

**pFunctionPointer_t**

> Does anyone think pFunctionPointer_t is more readable than Callback<void()>?

Also, I still advocate to typedef the `Callback` type (see above) and explicitly return a pointer, rather than typedef'ing the pointer type.

**Callback Priorities**

> Do you want your callback sorted or ordered ?

Yes, the callbacks are kept in the order in which they are `add`'ed or `add_front`'ed.  And for the most cases, the order is probably not important - different points in the application just want to make sure they have a chance to react to an event.  

However, I think inserting callbacks based on priority would be useful, and I don't think it adds that much complexity.  If we can agree on a "default" priority, say 1, then the following is all we really need:

```c++
class CallChainLink {
public:
    CallChainLink(): cb(), next(NULL), priority(1) {
        // No work to do
    }

    CallChainLink(Callback<void()> &callback): cb(callback), next(NULL), priority(1) {
        // No work to do
    }
    Callback<void()> cb;
    CallChainLink * next;
    int priority;
};

pFunctionPointer_t priority_add(Callback<void()> func, int priority = 1) {
    CallChainLink *new_link = new CallChainLink(func);
    if (NULL == _chain) {
        _chain = new_link;
        return &new_link->cb;
    }

    CallChainLink *link = _chain;
    while (true) {
        if (NULL == link->next) {
            link->next = new_link;
            return &new_link->cb;
        } else if (priority < link->next->priority) {
            new_link->next = link->next;
            link->next = new_link;
            return &new_link->cb;         
	}

        link = link->next;
    }
}
```

If we want a separate function for `add` and `priority_add`, then `add` is really just a wrapper for `priority_add` that passes the default priority.

ptpaterson, 2017-02-23:
A forked version of [`GenericCallChain.h`](https://gist.github.com/ptpaterson/7a1c88ca6fec9d85b75f3b7874003ced)

pan-, 2017-02-23:
@ptpaterson Thanks for your clarifications.

> The benefit to building in an aggregator into the CallChain would be to have call/() operator return the same value type as the Callbacks.

A generic aggregator would not always return a type identical to the one returned by the callbacks hosted by the callchain. Providing an implementation which cover this case is more complex and would add another template parameter to the template class and the creation of dedicated iterator to iterate over `CallChainLink` instances of the `Callchain`.  I would prefer to have this outside the `Callchain` which is really just a glorified single linked list of `Callback`.

> Also, I still advocate to typedef the Callback type (see above) and explicitly return a pointer, rather than typedef'ing the pointer type.

I agree, in that context it would also be interesting to allocate the Node with `new(std::nothrow)` rather than the regular `new` which will cause an abort if there is not enough memory to allocate the `Callback` node.

> Callback Priorities

If it is just about sorting Callbacks with an integral value I think it is fine. On the other hand, if it is about a more general sort implementation then I'm not sure that adding a new template parameter for the Comparison function object (like it's done for ordered container in the STL) is what we want in term of user friendliness.

pan-, 2017-02-23:
> It would then be difficult to extract the return value of Callback::call, which is as I say a big reason to otherwise build in an aggregator.

@ptpaterson It is easier than you may think. The simplest solution add a `ReturnType` typedef in the `Callback` class but a trait extracting the type is trivial: 

```c++
template<typename T>
struct identity {
	typedef T result;
};

// return the result type of a function type: R(A1,A2,...)
template<typename T>
struct get_callback_result_t;

// pattern matching  
template<typename R>
struct get_callback_result_t<R()> : identity<R> { };

template<typename R, typename A0>
struct get_callback_result_t<R(A0)> : identity<R> { };

template<typename R, typename A0, typename A1>
struct get_callback_result_t<R(A0, A1)> : identity<R> { };

// continue pattern matching for remaining arities

template<typename T>
class GenericCallchain { 

    typedef get_callback_result_t<T>::result ResultType;

// ...
}
```

geky, 2017-02-23:
Perhaps a prioritized version should be split into a separate class? Maybe PriorityCallChain? This would match similar design patterns for Queue and PriorityQueue.

I'm still not really sold on typedefing the callback type. The callback class itself has a simpler name than function pointers and including the type information in the name discourages users from only supporting a single callback type (such as what has happend in CallChain).
``` cpp
Callback_t *add(Callback_t func) { ... }
//vs
Callback<R(A0)> *add(Callback<R(A0)> func) { ... }
```

> There's no need to define a template for every arity out there

Good point! Though adding the template would still require a name change.

ptpaterson, 2017-02-24:
> I'm still not really sold on typedefing the callback type.

Wrote out a [full-ish comparison](https://gist.github.com/ptpaterson/d223669547d11d71f38ebf6041436d0c) of what it would look like in either case - no `Callback` typedefs vs typedefs... here's what I currently think:

**NOT Typedefing**
- PROS:  Being explicit about the Callback types is more indeed more clear and conveys intent better to a user/application.  
- CONS:  If a change is needed to the callback prototype, there will be more refactoring necessary.

**YES Typedefing**
- PROS:  Less typing up front.  Makes sense when you are the one writing the library.
-  CONS:  Not as clear, more looking around is necessary to determine what the type actually is.  This was not really an issue with `pFunctionPointer_t` because it could always be assumed `void()`.  Didn't actually save much time.

**Conclusion**
@geky has pulled me into his camp, and I now advocate for eliminating the typedef of `Callback`.

[Updated GenericCallChain.h](https://gist.github.com/ptpaterson/7a1c88ca6fec9d85b75f3b7874003ced)

ptpaterson, 2017-02-24:
> A generic aggregator would not always return a type identical to the one returned by the callbacks

Thanks @pan, I got a bit hung up on it.  I quite satisfied now with running some aggregator externally from the `CallChain`.  And my satisfaction is what is most important :D

pan-, 2017-02-24:
@geky @ptpaterson I believe having a typedef defining the type of `Callback` used is still important, even if it not used in the signature, it can help generic code to retrieve the type of `Callback` objects needed. For instance `Container::value_type` is not used to defined signature of insertion functions in the standard library but it can be used by generic code to retrieve the `value_type` held. 

About priorities, I though a bit more about it and operations like `add` or `add_front` doesn't map well (in my opinion) with the concept of a strictly ordered container. Two classes `Callchain` and `OrderedCallchain` 
with different contract for insertion operations would be more correct.

closed 2017-10-27 by ChiefBureaucraticOfficer