ARMmbed/mbed-os #3825: osMessageCreate will return NULL when enable DMBED_HEAP_STATS_ENABLED

kyliuxing, 2017-02-23:
mbed OS version : 5.1.5
compile cmd: mbed compile --source TESTS/TARGET_RDA/test/ --source ./ -DMBED_HEAP_STATS_ENABLED

I make a wrapper of osMessageCreate, when i define MBED_HEAP_STATS_ENABLED，osMessageCreate will return NULL. Is it a bug and has been fix in latest mbed version?

Thank you!
test case:
main.cpp
#include "mbed.h"
#include "rtos.h"

#define osWaitForever 0xFFFFFFFF
void* rda_msgQ_create(unsigned int queuesz)
{
void* internal_data;
osMessageQDef_t msgQ;
osMessageQId msgQId;

internal_data = (void *)malloc((4 + queuesz) * sizeof(unsigned int));**//Is it the root cause?**
msgQ.queue_sz = queuesz;
msgQ.pool = internal_data;
msgQId = osMessageCreate(&msgQ, NULL);

printf("msgQ addr %p\r\n", msgQId);
return (void *)msgQId;
}

int main() {
void *msgQ;
msgQ = rda_msgQ_create(4);
while (true) {
}
}
rt_cmsis.c
osMessageQId svcMessageCreate (const osMessageQDef_t *queue_def, osThreadId thread_id) {

if ((queue_def == NULL) ||
(queue_def->queue_sz == 0U) ||
(queue_def->pool == NULL)) {
sysThreadError(osErrorParameter);
return NULL;
}

if (((P_MCB)queue_def->pool)->cb_type != 0U) {
sysThreadError(osErrorParameter);
return NULL;
}

rt_mbx_init(queue_def->pool, (uint16_t)(4U*(queue_def->queue_sz + 4U)));
mbed_error_printf("\r\nsvc msg addr %p\r\n", queue_def->pool);//I add this to debug
return queue_def->pool;
}
Result:
MBED_HEAP_STATS_ENABLED is defined
svc msg addr 00100224
msgQ addr 00000000
MBED_HEAP_STATS_ENABLED is not defined
svc msg addr 001001f4 //it seems have two msg Queue
svc msg addr 00180018
msgQ addr 00180018

0xc0170, 2017-02-23:
cc @c1728p9 @bulislaw

kyliuxing, 2017-02-24:
@bulislaw 
It works!Thank you!
But the check is still enabled, how it was effected by macro DMBED_HEAP_STATS_ENABLED ?

bulislaw, 2017-02-24:
I've tried it on mbed OS master and it wasn't, it worked ok, but only the first time, when I've tried to allocate multiple queues then I could see NULL popping out. You generally can't rely on initial values of memory, I'm guessing 'luck' and little shift in memory access patterns shifted it around.
I'll close the issue now, please reopen it if you need further assistance.

closed 2017-02-24 by bulislaw