ARMmbed/mbed-os #3826: STM32F103 Nucleo CAN init hangs

vargham, 2017-02-23:
### Description
- Type: Bug
- Priority: Blocker

Can not use CAN bus with STM32F103 Nucleo. The init hangs. I used the online IDE with the newest mbed lib (136). Also tried the same code with Nucleo F302 and with F429, and there is no problem.
My test program below. The LED never blinks.
```
#include "mbed.h"

CAN can(PA_11, PA_12);
DigitalOut led1(LED1);
Ticker aliveTicker;

void alive()
{
    led1 = !led1;
}

int main()
{
    aliveTicker.attach(&alive, 1.0f);
    while (true)
    {

    }
}
```
I exported the project to uVision, and find that the CAN init calls can_frequency function and there is a never ending loop in can_api.c line 180.
`       while((can->MSR & CAN_MSR_INAK) == CAN_MSR_INAK)`

---------------------------------------------------------------
## Bug

**Target**
STM32F103

**Toolchain:**
Online IDE

**Toolchain version:**
mbed lib rev 136

**Expected behavior**
Init CAN interface without error.

**Actual behavior**
MCU hangs on CAN interface init.

**Steps to reproduce**
Try my simple example code above with target Nucleo F103

bcostm, 2017-02-23:
Hi,

I think your issue is due to the wrong configuration of the CAN_RD pins in the PeripheralPins.c file:

```
const PinMap PinMap_CAN_RD[] = {
    {PA_11, CAN_1, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)},
    {PB_8 , CAN_1, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 10)}, // Remap CAN_RX to PB_8
    {NC,    NC,    0}
};
```
These pins must be set as INPUT and not AF.

I have already corrected this in the PR #3737 and just checked your code using master branch and it works fine.

vargham, 2017-02-23:
Working. Thanks.

bcostm, 2017-03-01:
Issue to be closed. Thanks.

closed 2017-03-01 by 0xc0170