ARMmbed/mbed-os #3834: NRF51_DK error on test case: 'SPI - SD card exists'

TsungtaWu, 2017-02-24:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NRF51_DK

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
1.0.0

**Expected behavior**
In CI test, K64F is able to pass test case 'SPI - SD card exists'  

**Actual behavior**
NRF51_DK have ERROR on test case 'SPI - SD card exists'  

**Steps to reproduce**
Insert SD card on CI shield V1.0.0 and mount on NRF51_DK platform
goto ci-test-shield folder, then type
$ mbed test -m NRF51_DK -t GCC_ARM -n tests-api-spi* --app-config mbed_app.json

0xc0170, 2017-02-24:
cc @anangl @nvlsianpu

anangl, 2017-03-02:
There are two problems that cause these tests to fail. One, related to Nordic's HAL implementation, should be fixed by PR #3842. But the other one is that these tests need a lot of stack space (over 3 kB), due to the way the FATFileSystem is implemented. And currently, the stack size for the main thread is set to 2 kB for TARGET_MCU_NRF51822 (see [here](https://github.com/ARMmbed/mbed-os/blob/master/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h#L357)) and there is no an easy/elegant way to change it to make these tests pass. Some related discussion can be found in [#3315](https://github.com/ARMmbed/mbed-os/issues/3315) and PR #3675. I think we need to wait for CMSIS/RTX2 update mentioned [here](https://github.com/ARMmbed/mbed-os/pull/3675#issuecomment-276638184).

closed 2017-10-27 by ChiefBureaucraticOfficer