ARMmbed/mbed-os #3835: Xdot Lora

naushad-rahman, 2017-02-24:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question

I am trying to implement Callback function , Example sited by the mbed 

#include "mbed.h"

Mutex stdio_mutex;
Thread t2;
Thread t3;
    
void notify(const char* name, int state) {
    stdio_mutex.lock();
    printf("%s: %d\n\r", name, state);
    stdio_mutex.unlock();
}

void test_thread(void const *args) {
    while (true) {
        notify((const char*)args, 0); wait(1);
        notify((const char*)args, 1); wait(1);
    }
}

int main() {
    t2.start(callback(test_thread, (void *)"Th 2"));
    t3.start(callback(test_thread, (void *)"Th 3"));

    test_thread((void *)"Th 1");
}

- Related issue: 
I am getting Hard fault 
output is 
Hard FaulTh 1: 0
Th 2: 0
Th 3: 0

- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
XDot

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2017-02-24:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-02-27:
Not an ST device...

naushad-rahman, 2017-02-27:
It is ST device 

https://developer.mbed.org/platforms/MTS-xDot-L151CC/

bcostm, 2017-02-27:
I meant not an ST platform...

0xc0170, 2017-02-27:
>  @0xc0170 please change label to devices: multitech

I'll do. First thing would be to identify this within the multitech target, and see if this issue applies to a common code base that would affect other targets as well, then this could apply to ST platforms. 
Sorry for the initial confusion

0xc0170, 2017-02-27:
cc @mfiore02

janjongboom, 2017-02-27:
Is this on mbed OS 5.3.5? If so, downgrading to 5.3.1 might work due to #3823.

mfiore02, 2017-02-28:
@naushad-rahman @janjongboom @0xc0170 quick update...

I have a NUCLEO_L152RE on hand and this app looks OK on that platform with mbed-os-5.3.6.

I tested this app on the xDot starting with mbed-os-5.3.6.

Versions 5.3.6 - 5.3.1 all had the same behavior.
```
Th 1: 0
Hard FaulTh 1: 0
Hard FaulTh 1: 0
Hard FaulTh 1: 0
Hard FaulTh 1: 0
Hard FaulTh 1: 0
```
Hard faults ad infinitum.

Versions 5.3.0 - 5.2.0 also all had the same behavior.
```
Th 1: 0
```
No output after the first line.

I can't go back any further since Callback was introduced in mbed-os-5.2.0.

Debugging time! I'll post again when I know more.

mfiore02, 2017-02-28:
@janjongboom based on the progress we've made with #3823 I don't think this issue is related.

mfiore02, 2017-02-28:
It looks like the error is encountered during the context switch. Below is the stack trace and register info from the app compiled with mbed-os-5.3.6.

```
Breakpoint 1, HardFault_Handler ()
    at ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/TARGET_XDOT_L151CC/device/system_stm32l1xx.c:624
624	  printf("Hard Fault\n");
(gdb) bt
#0  HardFault_Handler ()
    at ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/TARGET_XDOT_L151CC/device/system_stm32l1xx.c:624
#1  <signal handler called>
#2  0x20001ec8 in ?? ()
#3  0x08003786 in rt_switch_req (p_new=0x20001a74 <mp_tcb+140>) at ./mbed-os/rtos/rtx/TARGET_CORTEX_M/rt_Task.c:106
#4  0x080035ce in rt_systick () at ./mbed-os/rtos/rtx/TARGET_CORTEX_M/rt_System.c:310
#5  0x080015ec in SysTick_Handler () at HAL_CM3.S:315
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

(gdb) info registers
r0             0x0	0
r1             0x2000163e	536876606
r2             0x0	0
r3             0x20001ec8	536878792
r4             0x20001ca8	536878248
r5             0x0	0
r6             0x20000168	536871272
r7             0x8007fa5	134250405
r8             0x20000168	536871272
r9             0x20000168	536871272
r10            0x2	2
r11            0x800d706	134272774
r12            0x3f9	1017
sp             0x20007f68	0x20007f68
lr             0xfffffff1	-15
pc             0x80041c2	0x80041c2 <HardFault_Handler+2>
xPSR           0x21000003	553648131
```

Entry 2 in the stack trace is in the processor's RAM, not FLASH. The ?? is also suspicious to me. I'm not a RTOS master, so I'm asking those who are: does the RTOS have components that execute out of RAM instead of FLASH? Or are we having some pointer issues?

klaas019, 2017-10-24:
Works as of mbed-os-5.5.7

closed 2017-11-15 by maclobdell