ARMmbed/mbed-os #3838: Thread::max_stack() return bad value

nvlsianpu, 2017-02-24:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Major

Thread::max_stack()  return lowest unused address from thread's stack instead of maximum of stack usage.
---------------------------------------------------------------
## Bug

**Target**
All

**Toolchain:**
All

**meed-os sha:**
https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5
3d4155e2ee1272bcb330692869c1d7a56b2625a4

nvlsianpu, 2017-02-24:
@pan-

pan-, 2017-02-24:
@bulislaw The bug is, I believe [here](https://github.com/ARMmbed/mbed-os/blob/feature_cmsis5/rtos/Thread.cpp#L225). size should be equal to `high_mark * sizeof(uint32_t*)`.

nvlsianpu, 2017-02-24:
size = (~~(uint32_t)thread->stack_mem +~~ thread->stack_size) - (high_mark * 4);

bulislaw, 2017-02-28:
I'll commit the fix soon:
```
diff --git a/rtos/Thread.cpp b/rtos/Thread.cpp
index 428fc2c5b..92cb2e340 100644
--- a/rtos/Thread.cpp
+++ b/rtos/Thread.cpp
@@ -222,7 +222,7 @@ uint32_t Thread::max_stack() {
         uint32_t high_mark = 0;
         while (((uint32_t *)(thread->stack_mem))[high_mark] == 0xE25A2EA5)
             high_mark++;
-        size = ((uint32_t)thread->stack_mem + thread->stack_size) - (high_mark * 4);
+        size = thread->stack_size - (high_mark * sizeof(uint32_t));
     }
 
     _mutex.unlock();
```

bulislaw, 2017-02-28:
Fixed now.

closed 2017-02-28 by bulislaw