ARMmbed/mbed-os #3845: Should semihosting be allowed from mbed-events callbacks?

janjongboom, 2017-02-27:
### Description
- Type: Question

## Question

https://forums.mbed.com/t/events-semihosting/2237

> I really am loving the API for the mbed-os Events; one thing though, it seems like one cannot safely do semihosting printfs from within the callbacks.
> 
> When I attempt this on a STM32F303K8 nucleo, the device will lock up within about 1-3 minutes
>
> Is this expected behavior? I've had to go direct to USB before for debug, which is no problem if that's a requirement.

0xc0170, 2017-02-28:
It should work. I wonder if we can get more details how it locks up (where exactly) and if we can get simple application/test to reproduce it easily.

cc @bcostm @adustm @LMESTM @jeromecoutant  
@geky

LMESTM, 2017-07-03:
@janjongboom @0xc0170 
F303K9 doesn't support MBED5, so I have run the below test on F091RC, F446RE and L476RG.
All run fine for 20 minutes and mode.  Maybe things have been fixed since report (there has been a fix on tickers) or the test code is different.  So please either provide a way to reproduce otherwise let's close it.
```
#include "mbed.h"
#include "rtos.h"

#define TEST_EQUEUE_SIZE 1024
EventQueue queue(TEST_EQUEUE_SIZE);
DigitalOut led1(LED1);

static int counter = 0;

void blink()
{
    counter++;
    //printf("Toggling stuff, counter=%d\r\n", counter);
    led1 = !led1;
    queue.call_in(250, blink);
}

int main() {
    printf("Toggling stuff test\r\n");
    queue.call_in(250, blink);
    queue.dispatch();
    while(1){
    }
}
```

LMESTM, 2017-07-03:
ST_TO_BE_CLOSED

closed 2017-07-03 by 0xc0170