ARMmbed/mbed-os #3847: Cannot clone mbed-OS in Windows if PATH length is already too long

SeppoTakalo, 2017-02-28:
Within mbed-OS longest filename is 165 characters.
Windows API allows only 260 characters.
https://msdn.microsoft.com/en-us/library/aa365247(VS.85).aspx#maxpath

|length| file name|
|------|----------|
|155|`./mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_STM32F4/TARGET_RELEASE/TARGET_M4/libconfiguration_stm32_cortex_m4_0x10000000_0x0.a`|
|157|`./mbed-os/features/nanostack/FEATURE_LOWPAN_BORDER_ROUTER/TOOLCHAIN_GCC/TARGET_LIKE_CORTEX_M0/libnanostack_arm-none-eabi-gcc_Cortex-M0_lowpan_border_router.a`|
|157|`./mbed-os/features/nanostack/FEATURE_THREAD_BORDER_ROUTER/TOOLCHAIN_GCC/TARGET_LIKE_CORTEX_M0/libnanostack_arm-none-eabi-gcc_Cortex-M0_thread_border_router.a`|
|163|`./mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_ARM_BEETLE_SOC/TARGET_DEBUG/TARGET_M3/libconfiguration_beetle_cortex_m3_0x20000000_0x140.a`|
|165|`./mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_ARM_BEETLE_SOC/TARGET_RELEASE/TARGET_M3/libconfiguration_beetle_cortex_m3_0x20000000_0x140.a`|

So cloning the application `mbed-os-example-mesh-minimal` requires already 193 characters.

Should we limit the PATH lengths for mbed-OS to allow more flexibility for users? We are now forcing users to work nearly the root folder of `C:\`

Also, this is a problem of some CI jobs, because they use `C:\WB\<job name>\<branch name>` which easily is close to 100 characters.

SeppoTakalo, 2017-02-28:
Please review: @sg- @mikter

jupe, 2017-02-28:
this has been problematic with CI as well

SeppoTakalo, 2017-03-14:
We need to raise the priority of this one.
If we cannot build mbed OS inside your Windows home directory, it is a bit limiting.

Also in CI the workspace root directory will be at least 54 characters long. Which leaves ~100 characters for job name, application name and branch name.

adustm, 2017-03-14:
Hello, This issue has been raised also here : https://github.com/ARMmbed/mbed-os/pull/3649
Someone should rename every directories.

0xc0170, 2017-03-14:
cc @theotherjimmy Let's discuss this topic this week.

SeppoTakalo, 2017-03-14:
I could save roughly 50 characters by removing `_arm-none-eabi-gcc_Cortex-M0_thread_border_router` from the library name. Basically it is already exposed in the folder structure.
But for that, I need to modify our build scripts.
I'll suggest that Uvisor team would do the same. `libconfiguration_beetle_cortex_m3_0x20000000_0x140.a` is 53 characters. The `TARGET_M3` is already in path name, also `BEETLE` is there. The `0x20000000_0x140` is too much information for library name, I would say.

adbridge, 2017-03-14:
@SeppoTakalo @mazimkhan I think this is an excellent idea Seppo, I see no point in the excessively long names being used there. This is a difficult overall problem as we have little control over how partners decide to structure / nest their targets for example. We need to think through how mbed-os is structured in its entirety to fix this in a semi future proof way...

adbridge, 2017-03-14:
@adustm Thanks for your renaming suggestions, we have considered such things. The problem we have is that the tools currently rely on certain naming patterns to the directory structures, thus any changes have to be made in conjunction with tool updates.

adustm, 2017-03-14:
> thus any changes have to be made in conjunction with tool updates.
@adbridge I know, I just wanted to centralize the information here :)

closed 2017-10-27 by ChiefBureaucraticOfficer