ARMmbed/mbed-os #3849: Error on SPI freq lower than requested one

LMESTM, 2017-02-28:
As reported here: https://github.com/ARMmbed/mbed-os/pull/3759#issuecomment-282786963

With this patch 100KHz no longer works on the Odin (NUCLEO_F439ZI) and NUCLEO_F429ZI. When the SDBlockDevice is used I get:
"Couldn't set suitable SPI freq: request:100000, lowest:328125"

c1728p9, 2017-02-28:
100KHz isn't an unreasonable frequency. Can the clock divisor code be updated to divide this low? I think there is a prescaler register (divide up to 256 in powers of 2) and a divider register (divide up to 255).

LMESTM, 2017-04-27:
I propose to close this issue for now as corresponding PR was merged.
If issues are reported we'll open a new one.

closed 2017-04-27 by LMESTM