ARMmbed/mbed-os #3851: Maxim platforms need exporter (project generation) support for common IDEs

maclobdell, 2017-02-28:
### Description
- Type:  Enhancement
- Related issue: N/A
- Priority: Major 

---------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

Maxim platforms (MAX32600MBED, MAX32620HSP, MAX32625MBED, MAX32625NEXPAQ, MAX32630FTHR, MAXWSNENV) have generic eclipse and makefile export support, but lack support for Keil uVision and IAR.  

**Suggested enhancement**

I highly recommend adding support for exporting Keil uVision and IAR, as these are very popular IDEs.

Information on how to do so can be found here:
https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/exporters/#adding-export-support-for-a-target

**Pros**
  Offer development environments with single-step debugging and advanced features. 

**Cons**
  None
-----------------------------------------------------------------

maclobdell, 2017-02-28:
cc @jessexm,  @kgills

0xc0170, 2017-03-07:
> cc @jessexm, @kgills

Any update?

closed 2017-10-27 by ChiefBureaucraticOfficer