ARMmbed/mbed-os #3853: [UBLOX_EVK_ODIN_W2] Default uVision settings fail on flash/debug

bridadan, 2017-02-28:
**I include a fix at the bottom, please read!**

### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**IDE:**
uVision

**IDE version:**
5.16a 

**mbed-cli version:** 
`1.0.0`

**meed-os sha:**
`3a27568a505bbc0bb8eeb973109d3d39ce823d1c`

**STLink version:**
Latest (`2.28.18`)

**Expected behavior**
Debugging and flashing work with default settings

**Actual behavior**
I get a "Unknown target connected" when starting the debugger/flasher.

**Steps to reproduce**
- Export any example to uVision and the UBLOX_EVK_ODIN_W2
    - `mbed export -m UBLOX_EVK_ODIN_W2 -i uvision`
- Compile
- Try to debug/flash

### Fix
If you go into the debug settings of the (correctly set) STLink debugger, changing the "Port" parameter from "JTAG" to "SW" fixed the issue.

cc @theotherjimmy

0xc0170, 2017-03-01:
> If you go into the debug settings of the (correctly set) STLink debugger, changing the "Port" parameter from "JTAG" to "SW" fixed the issue.

I had similar issue today when exported a project to IAR 7.8 (cmsis was set as jtag). took me a while to find this was causing flash failures. I can report this separately

bridadan, 2017-03-01:
As I mentioned in my original post, I was running a pretty old version of uVision (5.16a, from August 2015). As soon as I upgraded to 5.23 the issue went away. @0xc0170 I'd be ok closing this issue at this point if you're ok with that.

closed 2017-03-02 by 0xc0170