ARMmbed/mbed-os #3854: Build error occurs for deletion of last revision of FATFileSystem lib

TomoYamanaka, 2017-03-01:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

In [mbed offcial FATFileSystem library](https://developer.mbed.org/users/mbed_official/code/FATFileSystem/), version 3 or later seem to be gone.
History : https://developer.mbed.org/users/mbed_official/code/FATFileSystem/shortlog

FATFilesystem lib revision 9 existed at least Jan 23, 2017.
Example of revision 9 : https://developer.mbed.org/teams/Renesas/code/USBHost_custom/rev/f8c0979c2814

Therefore, for example, when importing the sample including USBHost library in last revision, it shows the following "Import Warning" message.
`Unable to update library to revision: /GR-PEACH_Camera_in/USBHost/FATFileSystem`
Then when building the sample, many build error like a type mismatch occurs due to no last revision of FATFileSystem library.

geky, 2017-03-01:
@0xc0170, @adbridge, @bridadan, @theotherjimmy 

I'm not sure how the online site synchronizes the mbed 2 repositories, but I suspect this has happened as a result of the FATFileSystem being moved out of the mbed 2 libraries and into the mbed OS 5 set of features.

The "Legacy Warning" is correct that users should switch to mbed OS 5 for the most recent version of the FATFileSystem, but to avoid breaking users code, does anyone know how to stop the mbed 2 FATFileSystem from updating?

adbridge, 2017-03-06:
For mbed 2 we only update the 'mbed' lib and the 'mbed-dev' lib on the mbed org site. If however, partners are using any of the code that sits in the 'features/unsupported' path of mbed-os themselves then that could change at any point... Which I suspect is what has happened at this time? But the features path is not in the 'mbed' lib anyway...

0xc0170, 2017-03-06:
As I understand, the FAT library on developer page had more revisions that it currently has.

Wait a sec, @TomoYamanaka isn't this library that your app is using - https://developer.mbed.org/teams/mbed-official/code/FATFileSystem/ - it contains  the 9th revision that you referenced above.

geky, 2017-03-06:
@adbridge, neither of us have modified the repo on the mbed org site I believe

TomoYamanaka, 2017-04-17:
How is this going ? Any update ?

TomoYamanaka, 2017-07-25:
I confirmed the fix by using not [above FATFileSystem](https://developer.mbed.org/users/mbed_official/code/FATFileSystem/), but also [this FATFileSystem](https://developer.mbed.org/teams/mbed-official/code/FATFileSystem/ ).
So I close this issue.

closed 2017-07-25 by TomoYamanaka