ARMmbed/mbed-os #3855: STM32F091: Peripherals keep running when debugging

markus-becker-tridonic-com, 2017-03-01:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32F0

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4.1 20160609 

**mbed-cli version:** 
0.9.10

**mbed-os sha:**
5.3.4 and earlier

**Expected behavior**
Peripherals get stopped when debugger is on a breakpoint.

**Actual behavior**
Peripherals (i.e. ADC) continue to deliver interrupts when debugger is on a breakpoint. This leads to an RTX FIFO overflow.

**Steps to reproduce**
In a separate thread enable ADC sampling, Set break point in main thread. If you try step debugging, at some point RTX FIFO overflow will occur.

@bcostm @MarceloSalazar

LMESTM, 2017-06-02:
@markus-becker-tridonic-com 
Hi, at the first glance, I wouldn't say this is a bug, and probably not a SW bug. 

The Peripherals are clocked asynchronously to the CPU core on which you're debugging SW. When you step over an instruction  for writing a byte over UART, you'd expect the character to go up to the remote device wouldn't you ? So even though the debugger will stop the SW after the execution of the register write access, the UART IP needs to be clocked anyway for the HW action to proceed.

Nevertheless there are few peripherals that you would like to stop when debugging, like watchdogs in particular. In the Debug Support chapter of STM32 reference manuals, you will find the list of few peripherals that can be stopped when debugging. For F091RC, in chapter 32.9.4 there is the list of peripherals that can be configured to stop some of the peripherals.  You can freeze Timer clock counter, I2C SMBUS timeout and watdchdogs using DBGMCU_APB1_FZ register. But ADC is not part of them.

In the case of your example, you need to let the main thread 'consume' the ADC samples, or maybe put in place a control mechanism between the consumer of ADC samples and the producer (ADC thread). I don't know how you SW works ...

LMESTM, 2017-06-13:
I think we have answered the question. I propose to close the issue, unless further questions arise.
ST_TO_BE_CLOSED

LMESTM, 2017-07-03:
@markus-becker-tridonic-com @0xc0170  let me know if anything can be done otherwise please feel free to close this issue

closed 2017-07-03 by 0xc0170