ARMmbed/mbed-os #3856: All lpcxpresso exports are broken: no instructions for GCC_CR in profiles

janjongboom, 2017-03-01:
Every export to lpcxpresso breaks with:

```
project.py: error: argument --profile: toolchain GCC_CR is not supported by profile ./mbed-os/tools/profiles/default.json
```

This is because none of our profiles have GCC_CR configurations in them.

@theotherjimmy

theotherjimmy, 2017-03-01:
See https://github.com/ARMmbed/mbed-os/blob/master/tools/export/lpcxpresso/__init__.py#L23

LPCXpresso does not use GCC_CR.

janjongboom, 2017-03-01:
@theotherjimmy then why do the exports whine about GCC_CR? /cc @andcor02

theotherjimmy, 2017-03-01:
@janjongboom They don't
```
$ mbed export -i lpcxpresso -m lpc1768  
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_BLE
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_NANOSTACK
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_STORAGE
```

Works for me.

janjongboom, 2017-03-02:
@theotherjimmy On mbed-os-example-blinky with mbed OS 5.3.6:

```
janjon01:~/repos/mbed-os-example-blinky (master) $ mbed export -i lpcxpresso -m lpc1768
WARNING: MBED_GCC_ARM_PATH set as environment variable but doesn't exist
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: argument --profile: toolchain GCC_CR is not supported by profile /Users/janjon01/repos/mbed-os-example-blinky/mbed-os/tools/profiles/debug.json
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u /Users/janjon01/repos/mbed-os-example-blinky/mbed-os/tools/project.py -i lpcxpresso -m lpc1768 --source ." in "/Users/janjon01/repos/mbed-os-example-blinky"
---
```

theotherjimmy, 2017-03-02:
https://github.com/ARMmbed/mbed-os/pull/3791 Fixed this. It's slated for 5.4.0.

janjongboom, 2017-03-02:
Great, thanks.

closed 2017-03-02 by janjongboom