ARMmbed/mbed-os #3858: Mbed lib : target inheritance appears to be broken

adbridge, 2017-03-01:
### Description
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
NRF51_MICROBIT_B
NRF51_DONGLE_LEGACY 

**Toolchain:**
ARM

When compiling a test application which has the mbed lib (mbed-os 2 sdk) included, the above two targets fail to compile with the following error:

DEBUG:root:Library name 'platform' is not unique (defined in '/src/mbed/TARGET_NRF51_MICROBIT_B/mbed_lib.json' and '/src/mbed/TARGET_NRF51_MICROBIT/mbed_lib.json')                      

DEBUG:root:Library name 'platform' is not unique (defined in '/src/mbed/TARGET_NRF51_DONGLE_LEGACY/mbed_lib.json' and '/src/mbed/TARGET_NRF51_DONGLE/mbed_lib.json')                     

It looks like the target inheritance is thus broken.

**Steps to reproduce**
Compile a version of blinky with mbed lib version 137 in the online IDE

0xc0170, 2017-03-02:
cc @theotherjimmy

theotherjimmy, 2017-03-02:
@adbridge @0xc0170 This is not a bug. look at the relevant definitions in targets.json.
```json
    "NRF51_DONGLE_LEGACY": {
        "inherits": ["MCU_NRF51_32K"],
        "extra_labels_add": ["NRF51_DONGLE"],
        "release_versions": ["2"],
        "device_name": "nRF51822_xxAA"
    },
    "NRF51_DONGLE_BOOT": {
        "inherits": ["MCU_NRF51_32K_BOOT"],
        "extra_labels_add": ["NRF51_DONGLE"],
        "macros_add": ["TARGET_NRF51_DONGLE"]
    },
    "NRF51_DONGLE_OTA": {
        "inherits": ["MCU_NRF51_32K_OTA"],
        "extra_labels_add": ["NRF51_DONGLE"],
        "macros_add": ["TARGET_NRF51_DONGLE"]
    },
    "NRF51_DONGLE": {
        "inherits": ["MCU_NRF51_32K_UNIFIED"],
        "progen": {"target": "nrf51-dongle"},
        "device_has": ["ERROR_PATTERN", "I2C", "I2C_ASYNCH", "INTERRUPTIN", "LOWPOWERTIMER", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SERIAL_ASYNCH", "SLEEP", "SPI", "SPI_ASYNCH", "SPISLAVE"],
        "release_versions": ["2", "5"]
    },
```
All of these are public targets, so of course there well be double mbed_lib.json's when building for release version 2.

adbridge, 2017-03-06:
As these are both compilable targets, then something is broken... 
@pan-

0xc0170, 2017-03-17:
@adbridge I recall there was an update to this issue?

adbridge, 2017-03-17:
It's waiting on @pan- to update the target entries, as they are incorrectly structured according to @screamerbg

theotherjimmy, 2017-04-25:
@adbridge Is this still a problem? could we close this issue?

adbridge, 2017-04-26:
@theotherjimmy There is still an issue with these target definitions that needs @pan- to fix. However  we can remove the tools label!

closed 2018-12-19 by ciarmcom