ARMmbed/mbed-os #3859: MAXIM targets fail to compile with mbed-dev (rev 137)

adbridge, 2017-03-01:
### Description
- Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
MAX32620HSP / MAXWSNENV

**Toolchain:**
ARM

**Toolchain version:**
Online IDE

**Steps to reproduce**
Compile a version of blinky with mbed-dev library for either of the above 2 targets in the online IDE. The following error is observed:
Error: L6218E: Undefined symbol __rt_heap_descriptor (referred from malloc.o).

0xc0170, 2017-03-17:
Is this still valid?

cc @kgills @jeremybrodt

adbridge, 2017-06-22:
@0xc0170 Yes neither of these targets build for mbed2 in the online IDE API, but only with mbed-dev source library. They build ok with the pre-compiled mbed binaries.

maclobdell, 2019-09-25:
closing as this is very old, release methods have changed, and this issue is not a priority any longer.

closed 2019-09-25 by maclobdell