ARMmbed/mbed-os #3865: [OOB_5_4_RC1] mbed CLI export to IAR breaks the debugger download for CMSIS DAP

senthilr, 2017-03-01:
# Description
When exporting IAR Workbench project files from mbed CLI, the IAR debugger download and run operation fails ("can't load emulator" error) to connect to target. 

# Environment 
- K64F
- IAR Embedded Workbench 
- Export from mbed CLI:
```
$ mbed ls
mbed-os-example-blinky (c0bf2d16e4c1)
`- mbed-os (3a27568a505b)
$ mbed --version
1.0.0
$mbed export -i iar
```
# Details & Workaround
Changing the debugger Interface option for CMSIS DAP from JTAG (default) to SWD eliminates the error and Download/Debug works as expected from the IDE.

Looking further at the template file for IAR [exporter](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.4/tools/export/iar/ewd.tmpl#L351) the default option is set for JTAG (vs CMSIS-DAP)

0xc0170, 2017-03-02:
Thanks for the report. I can confirm this is an issue.

@theotherjimmy

0xc0170, 2017-03-02:
@senthilr I can see you already provided a fix, but dont see it here in mbed-os. I can do it quickly and test it locally.

0xc0170, 2017-03-03:
Should be fixed, rc2 will contain this fix

closed 2017-03-03 by 0xc0170