ARMmbed/mbed-os #3873: Possibility to know how many events are in an eventqueue

daniel-cesarini-tridonic-com, 2017-03-02:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question
- Priority: Major

## Question

**Is it possible to know how many events are in an eventqueue?**

In particular, is it possible to have the current and maximum number of events present in an eventqueue (https://github.com/ARMmbed/mbed-os/tree/master/events)?

We have seen that in equeue_alloc code there is some checking of available memory, and we are thus wondering whether any functionality to obtain the above mentioned information is or would be available?

0xc0170, 2017-03-02:
What would be the use case? Can you share what drives this new addition?

geky, 2017-03-02:
Unfortunately not, the event queue supports dynamically sized contexts, which makes determining how many more events can be dispatched complicated. Even if you can dispatch many small events, fragmentation may prevent you from allocating a single larger event.

If you are only using fix-sized events you can just keep a counter tracking how many events have been dispatched.

I am curious though, what is the use case for this? Is this for sampling statistics?

daniel-cesarini-tridonic-com, 2017-03-03:
The use case is to tailor the EventQueue size to the actual maximum number of events that need to be handled in the queue at the same time (in order to get a statistic estimate, yes).

I am then wondering about the effect of this
```
// Creates an event queue enough buffer space for 32 Callbacks. This
// is the default if no argument was provided. Alternatively the size
// can just be specified in bytes.
EventQueue queue(32*EVENTS_EVENT_SIZE);
```
if you say that the allocation is dynamic, what is that for (by the way, on our codebase the allocation of such an EventQueue with 32 (that is the default) seems to use 2.3KByte of RAM. Is that right?

geky, 2017-03-03:
Adding this sort of statistics is a good idea, though we don't have any hooks in place currently. There is an internal variable that holds how much space is untouched (`EventQueue._equeue.slab.size`), but this may change from version to version, and a different implementation of the event queue may not have access to this information at all.

> if you say that the allocation is dynamic

Ah sorry, I should clarify. The events are variable size, but the memory the event queue operates out of is not. This is required to allocate events in interrupt context safely (allocations from the general purpose heap is not irq safe).

Variable sized events introduce fragmentation in this memory region, which makes it difficult to know exactly how many more events can be allocated.

``` cpp
EventQueue queue(8*sizeof(int)); // 8 words of storage
queue.call(func, 1);       // requires 2 words of storage
queue.call(func, 1, 2, 3); // requires 4 words of storage
// after this we have 2 words of storage left

queue.dispatch(); // free all pending events

queue.call(func, 1, 2, 3); // requires 4 words of storage
queue.call(func, 1, 2, 3); // fails
// remaining storage has been fragmented into 2 events with 2 words 
// of storage, no space is left for a 4 word event even though 4 bytes
// exist in the memory region
```

The event queue does have some strong garuntees on fragmentation. Once an allocation has been successful, that unit of memory will never be fragmented further (this is important for periodic systems). You could calculate how many events are available for a specific size of event, but this is also complicated by untouched space, which can service any event that fits.

daniel-cesarini-tridonic-com, 2017-03-06:
So in the following case:
```
EventQueue queue(8*sizeof(int)); // 8 words of storage
queue.call(func, 1);       // requires 2 words of storage
queue.call(func, 1);       // requires 2 words of storage
queue.call(func, 1);       // requires 2 words of storage
queue.call(func, 1);       // requires 2 words of storage
// after this we have 0 words of storage left

queue.dispatch();  // free all pending events
// now all memory should be free again (8 words)?

queue.call(func, 1, 2, 3); // requires 4 words of storage --> Would this fail now?
```
would that actually fail because of fragmentation?
and are at the end still 4 words free?

Is there any other documentation about the EventQueue memory pool organization/management?

geky, 2017-03-07:
> would that actually fail because of fragmentation?
Yes, it's an unfortunate side effect of how the allocator works. 4 words of storage are free, but only for allocations <2 words.

> Is there any other documentation about the EventQueue memory pool organization/management?
Not currently, but it does sound like there should be.

closed 2017-10-27 by ChiefBureaucraticOfficer