ARMmbed/mbed-os #3874: [OOB mbed-os-5.4] Blinky doesn't run when compiled with IAR for target MTS-mDot-F411

mray19027, 2017-03-02:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os-example-blinky/issues/45

Run fails for blinky when compiled with IAR for target MTS-mDot-F411

---------------------------------------------------------------
## Bug

**Target**
MTS-mDot-F411

**Toolchain:**
IAR

**Toolchain version:**
`IAR ANSI C/C++ Compiler V7.80.4.12462/W32 for ARM`

**mbed-cli version:** 
`1.0.0`

**meed-os sha:**
`3a27568a5 Merge pull request #3773 from geky/fs-filesystem-simple-3`

0xc0170, 2017-03-03:
This issue should fix it - https://github.com/ARMmbed/mbed-os/pull/3875 , should not allow this platform ot export , more details there

0xc0170, 2017-03-13:
Should be fixed via the referenced PR above

closed 2017-03-13 by 0xc0170