ARMmbed/mbed-os #3876: Deprecated `toolchain.h` warnings when compiling mbed TLS examples.

andresag01, 2017-03-03:
### Description
- Type: Bug
- Priority: Minor

When compiling any of the examples in mbed-os-example-tls, the following compilation warning is repeatedly printed:
```
[Warning] toolchain.h@23,0:  #1215-D: #warning directive: toolchain.h has been replaced by mbed_toolchain.h, please update to mbed_toolchain.h [since mbed-os-5.3]
```
It seems that the `toolchain.h` file has been renamed, but not all `#include` directives to it have been directed. I did a quick grep and found at least four of these, but perhaps there are more:

```
./benchmark/mbed-os/drivers/FileLike.h:19:#include "platform/toolchain.h"
./benchmark/mbed-os/features/FEATURE_BLE/source/BLE.cpp:30:#include <toolchain.h>
./benchmark/mbed-os/platform/platform.h:28:#include "platform/toolchain.h"
./benchmark/mbed-os/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/drivers/system/system.c:48:#include <toolchain.h>
```

I think that these `#include` preprocessor directives should be updated to `mbed_toolchain.h`.

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM and ARM

**Expected behavior**
Deprecated warnings should not be displayed during the compilation process

**Actual behavior**
The following compilation warning is printed many times:
```
[Warning] toolchain.h@23,0:  #1215-D: #warning directive: toolchain.h has been replaced by mbed_toolchain.h, please update to mbed_toolchain.h [since mbed-os-5.3]
```

**Steps to reproduce**
* Clone the mbed-os-example-tls repo
* Update the mbed-os.lib file to point to the mbed-os-5.4.0-rc1
* Deploy mbed-os
* Compile any of the examples for K64F using the following command:
```
mbed compile -t GCC_ARM -m K64F
```

0xc0170, 2017-03-03:
There's a patch inline, https://github.com/ARMmbed/mbed-os/pull/3868 that should resolve this. I'll review it now.

geky, 2017-03-03:
Is this still happening now that #3868 is merged?

closed 2017-03-03 by 0xc0170

noutram, 2017-06-27:
Same warning when exporting to uVision5