ARMmbed/mbed-os #3878: EEP/I2C WR 2 bytes ci-test FAIL on some STM32 platforms

bcostm, 2017-03-03:
### Description
- Type: Bug
- Related issue: #2957
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

For info, the test is OK on NUCLEO_F429ZI, NUCLEO_F412ZG, NUCLEO_F401RE
Seems related to the I2C IP version embedded in the MCU. To be checked.

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4 2016q3

**mbed-cli version:** 
(`mbed --version`)

**mbed-os:**
mbed-os-5.4

**Expected behavior**
Test is PASS

**Actual behavior**
Test if FAIL

**Steps to reproduce**
'mbed test' + ci-test shield

LMESTM, 2017-03-03:
@bcostm only on GCC_ARM or other toolchains as well ?
Just tested ok with ARM here ... 

+-------------------+---------------+---------------+--------+--------------------+-------------+
| target            | platform_name | test suite    | result | elapsed_time (sec) | copy_method |
+-------------------+---------------+---------------+--------+--------------------+-------------+
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | OK     | 13.81              | shell       |
+-------------------+---------------+---------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite    | test case                          | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.06               |
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+

LMESTM, 2017-03-03:
@bcostm - I think this is due to https://github.com/ARMmbed/ci-test-shield/issues/22
The I2C library needs to be updated, which I've done locally, but this hasn't been done in the ci-test-shield official repo yet ...

0xc0170, 2017-03-17:
This should be now resolved as ci-test-shield was fixed

closed 2017-03-17 by 0xc0170

bcostm, 2017-03-20:
Just for reference, I have verified the I2C tests and they are OK now.

```
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite    | test case                          | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.1                |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.1                |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.07               |
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
+-----------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite    | test case                          | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.04               |
+-----------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite    | test case                          | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.09               |
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.07               |
[... cut ...]
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-IAR | NUCLEO_L476RG | tests-api-i2c | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.06               |
+-------------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
```

ci-test-shield version:
53f8fcb Merge pull request #45 from jeromecoutant/PR_PIN

mbed-os version:
f4864dc Merge pull request #3930 from ARMmbed/release-candidate