ARMmbed/mbed-os #3883: Socket attach funciton bug

NirSonnenschein, 2017-03-05:
Type: bug
Rleated issue: not sure (possibly https://jira.arm.com/browse/IOTUPD-1238)
Priority: Blocker (blocking issue for 1.2 release)

We really appreciate your help on this issue as it is relevant to the 1.2 release (please feel free to forward/add people relevant to this issue).

## Bug
**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.2.1 20161205 (I think)

**mbed-cli version:** 
1.0.0

**meed-os sha:** (I tried several different versions from the 5.2 and 5.3 branches, with similar results
[Mbed_socket_attach_test.txt](https://github.com/ARMmbed/mbed-os/files/819467/Mbed_socket_attach_test.txt)
)
bcf7085 Merge pull request #3755 from ARMmbed/release-candidate

**Expected behavior**
Here is the basic scenario:
1.      Create Ethernet interface and connect it.
2.      Create new TCP socket
3.      Connect socket to www.arm.com at port 80
4.      Sent basic http request ("GET / HTTP/1.0\r\n\r\n") 
5.      Sleep for 20 seconds (osDelay)
6.      Call recv to receive 10 bytes of data.
During the sleep period we don’t see any callbacks (we have verified that the server response does indeed arrive during that time and recv still not block even if the sleep is for 3 seconds).

My expectation is when attaching a callback function to a socket is there when data arrives for this socket the callback function will be called asynchronously.
In my testing I have discovered that the callback for incoming data comes only after the recv function is called and will not arrive if the function is not called(tested on both blocking and non-blocking TCP sockets). 
Note: Kevin bracey suggested in a discussion with him regarding this issue that we may need to call recv once before the data arrives to get a callback (so this was added to the example, but behavior is similar).
Also Note:the code in quesiton uses the Ethernet interface (LWIP?), our port directly over Lwip code, does return a callback in this case.
Also Note:  kevin also suggested that Pull request 3619 might resolve the issue, however I tried merging these changes locally and still found the same behavior.  

**Actual behavior**
no callbacks prior to calling the recv function after the send. this doesn't allow us to use the callbacks as a way to detect that data has arrived (if we have to call recv to get it, there is not much point in a callback)

**Steps to reproduce**
I checked using the following simplified function:

void socketAttachSanity()
{
	printf ("start socket sanity");
	const char* message = "GET / HTTP/1.0\r\n\r\n";
	char buffer[100] = { 0 };
	nsapi_error_t status = 0;
	
	SocketAddress translatedAddress; 
	TCPSocket* socketObj = NULL;
	
	EthernetInterface* netInterface = new EthernetInterface();
	if (NULL != netInterface)
	{
		printf("new interface created\r\n");
		status  = netInterface->connect();
		if (NSAPI_ERROR_OK == status)
		{
			printf("interface connected : OK \r\n");
		}
		else {
			printf(" \r\nfailed to connect interface\r\n");
			return;
		}
	}
	else 
	{
		printf(" \r\nfailed to allocate interface\r\n");
		return;
	}
	
	printf("interface connected : OK \r\n");
	socketObj = new TCPSocket(netInterface);
	if (socketObj == NULL)
	{
		printf(" \r\nfailed to allocate socket\r\n");
		return;
	}
	socketObj->set_blocking(false); // also works for false
	socketObj->attach(socketCallback1);
	
	status =  socketObj->recv(buffer, 0); // also works if reading a positive size
	
	printf ("recv with status: %d ", status );
	
	status = socketObj->connect("www.arm.com", 80);
		
	if (status != NSAPI_ERROR_OK)
	{
		printf(" \r\nfailed to allocate socket\r\n");
		return;
	}
	status = socketObj->send(message,strlen(message) );
	printf ("send with status: %d ", status );
	if (status < NSAPI_ERROR_OK)
	{
		printf(" \r\nfailed to send socket\r\n");
		return;
	}
		
	printf ("\r\nsleeeep\r\n" );
	osDelay(20000); // also works for busy wait instead of sleep
	printf ("\r\after sleep\r\n" );
	
	status =  socketObj->recv(buffer, 10);
	printf ("recv with status: %d ", status );
	if (status < NSAPI_ERROR_OK)
	{
		printf(" \r\nfailed to recv socket\r\n");
		return;
	}
		
	//cleanup
	socketObj->close();
	delete socketObj;
	netInterface->disconnect();
	delete netInterface;
	printf ("end socket sanity ");
}

0xc0170, 2017-03-05:
cc @geky @kjbracey-arm

kjbracey-arm, 2017-03-06:
(@JanneKiiskila )

@mikaleppanan and I were unable to reproduce this, as discussed via e-mail.

With lwip and your test program, we did get a callback after the send.

If you were to simply issue a read call when you get a callback, it will work.

Can you show an actual program which reads when it gets a callback failing?

mikaleppanen, 2017-03-06:
Here are the debug prints I got from that example program. "application callback() !!!!!" is traced when callback is called by socket interface.

------------
Starting mbed Client example in IPv4 mode
[EasyConnect] Using Ethernet
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 192.168.100.22
[EasyConnect] MAC address d2:47:62:a3:42:84
start socket sanityinterface connected : OK 
recv with status: -3001 
 application callback() !!!!!
send with status: 18 
sleeeep

 application callback() !!!!!
fter sleep

 application callback() !!!!!
------------------

NirSonnenschein, 2017-03-06:
for my runs I never got a callback between "seeep" and "after sleep"
it seems to work better on your side than it does on mine, can you please specify the version fo MbedOS you used for testing so that I can re-check?

mikaleppanen, 2017-03-06:
This is the latest master of mbed-os.

kjbracey-arm, 2017-03-06:
Don't forget there is a potential race here - the callback could very well be happening before "sleeeep" in your setup. Changing trace may be affecting the precise timing.

And in principle, a callback could even happen before the write(), if the server decided to send before you wrote (unlikely, but could happen if app specified an incorrect port so you weren't talking to an HTTP server), or if it closed the connection on you before you wrote (quite possible).

So to correct myself above - "read when you get a callback" should work in this case, but isn't 100%, given the potentially edge-triggered nature of the event. Correct code for handling potentially edge-triggered events, as I've said before:

# Read from the socket (or write - same logic applies)
# If call works, process result and go to 1. If error conditions other than EWOULDBLOCK, deal with it.
# If the socket call said EWOULDBLOCK, you are allowed to go to sleep until you get a SIGIO. Then go to 1.

Just the same way you'd write an interrupt handler for the serial port, for example. That also wouldn't work reliably if you waited for a serial interrupt before each read attempt - it would fail if the serial port was already readable, and serial interrupts were edge-triggered (which they often are). You have to drain the port to force a "has become readable" interrupt edge. Potentially the same for sockets, depending on the underlying stack.

NirSonnenschein, 2017-03-06:
Hi, 
I've re-checked and this issue does indeed seem to be fixed in the mbedOS master (the 5.4 branch seems to contain the fix as well), however the problem is still present in the latest 5.3 branch. the , so it is probably a recent fix.
this still presents an issues creating code which will work reliably on 5.3 and 5.4 and later, is it possible to add the fix to 5.3?

regarding the callback happening early (I did a bit of testing and there is some latency before the server answers so this is not likely) also note that the same code exhibits the problem on 5.3 but not on 5.4 and latet so the problem is likely to be elsewhere.

as the code presented in the example is just a simple scenario designed to reproduce the issue, we are checking how to implement proposed changes to the actual scenario:
1. read after every callback. (would reading 0 bytes be sufficent? MedOS does seem to support this, and will return would block if there is no data to read)
2. try to read before getting a callback to see if we get would block (is reading 0 bytes ok here as well)? (otherwise we can be sure the socket is readable and there is not need to wait for the callback ).

kjbracey-arm, 2017-03-06:
> this still presents an issues creating code which will work reliably on 5.3 and 5.4 and later, is it possible to add the fix to 5.3?

Maybe if we knew what the fix was - you said it wasn't #3619 , and I don't know what else would have affected behaviour here. And wasn't #3619 already backported to 5.3.y? Maybe it wasn't because of the comment there that a change might upset some callback-pattern-sensitive users?

At the moment I think you may be just suffering from random timing effects, rather than an actual fix having occurred.

> regarding the callback happening early (I did a bit of testing and there is some latency before the server answers so this is not likely) 

Not in normal use, but imagine if you were single-stepping through the code, or the system was heavily loaded. Robust code wouldn't depend on "likely" timing behaviour.

> try to read before getting a callback to see if we get would block 

That's pretty close - read whenever you want more data. Which in this case is immediately after the write.

Only when a read says "would block" do you wait for a callback before trying again. (You could retry as often as you want, but the callback is there to allow you to limit the retries).  The "wait for callback" should always be a response to "non-blocking network call not ready". Never assume that a network call won't work until you get a callback - try it and see.

> (is reading 0 bytes ok here as well)? 

There shouldn't be any need for a separate "read 0". That particular call falls into a slightly fuzzy area - it might be a total no-op, or it might throw away UDP datagrams. It's not a viable readability check.

If it was generally possible to losslessly determine whether a socket was readable, the abstraction layer could provide that as API, but it isn't. All known stacks provide non-blocking read and write calls, but they don't all provide non-destructive "pre-check"s.

But you don't need to pre-check if the socket is readable. Just proceed to do your actual read, and see if you get "EWOULDBLOCK".

If you've got a state machine, presumably after issuing your write, you go into some sort of "data pump" state. And that presumably attempts a read on receipt of a callback. Make it do the same thing on initial entry to the state. Don't wait for an initial callback. And make sure it reads as much as possible before waiting for another callback.

kjbracey-arm, 2017-03-06:
Update:

Mika should be able to confirm this tomorrow, but current hypothesis for the difference is that 5.3 with lwIP is tending to very reliably generate an unnecessary callback inside every write() call. And because your code apparently reliably ignores such a callback, no more are generated - NSAPI assumes you're not interested, because you're not making socket calls.

In 5.4, there are fewer unnecessary callbacks from lwIP, and that particular "callback on every write" doesn't happen. So you don't end up consistently ignoring a callback, so you do get a later callback.

If that is what it is, backporting that change (#3754) to 5.3 might fix your code which incorrectly relies on that spurious callback not happening, but could just as well break other code that is incorrectly relying on it happening. So I don't think it's necessarily a net win for a backport.

To avoid race conditions, you must handle the case of callback before write() returns, and it appears 5.3 is demonstrating reliably that you don't, thanks to a spurious callback triggered by the write. (You must also handle spurious callbacks in general).

Simplest fix is the approach already suggested - always attempt your first read() as soon as you want data - immediately after write().

mikaleppanen, 2017-03-07:
Yes, the change between 5.3 and 5.4 is caused by event filtering done in: #3754. As Kevin stated it changed the timing so that after write() the application event is sent later.

NirSonnenschein, 2017-03-07:
OK, I think the basic behavior is problematic (having to read from a socket an extra time because of a spurious "send success" event  is problematic especially since it is not cearly difined what triggers an event) but we will try to to change our implementation accordingly (I think we can make this work for the specific use case).

avolinski, 2017-03-07:
Hi,

I think that no matter what should be the behavior from the underlying network stack.
it should be well defined for the user. Because user needs to know exactly what to expect, so he can implement his code.
User knows what to expect if he would use directly the underlying network stack (such as lwip)
But since it is being abstracted by mbedOS, which is very good, then the responsibility falls on the mbedOS networks abstraction mechanism to behave in the same way for every stack it abstracts.
and if mbedOS alters the expected behavior, it needs to be certainly well documented,
what is the new behavior.  
So again, user will know for what to prepare. And if the behavior from mbedOS is different from stack to stack, then user needs to be able to query on which stack he is working now, and adapt his code with branching logic for every stack differently. But then i see less logic in abstracting network stacks by mbedOS, their functionality can be as well exposed directly to user.

kjbracey-arm, 2017-03-07:
> having to read from a socket an extra time because of a spurious "send success" event is problematic especially since it is not cearly difined what triggers an event

You have to read whenever you want data, to avoid all sorts of pitfalls due to race conditions on edge-based signals. By not doing that, you're code is still liable to fail, despite it currently appearing to work.

You aren't demanding a signal before trying the connect or the send. So why are you demanding a signal before trying the read? That makes no logical sense - there's no consistent pattern to your use of the signal. You're inventing assumptions about what it means and when it might be delivered.

You need to get out of the mindset of trying to drive your code through the signal.

Simply treat SIGIO as a /hint/ about it being a good time to /retry/ socket calls.  The socket calls are the things that give you information, not the signals. The results of the socket calls advance your state machine, not the signals.

It is clearly defined that you will get at least one SIGIO to prompt you to retry after EWOULDBLOCK, and that is sufficient to make those non-blocking calls useful, without any need for a retry timer.

Write the code as if it was blocking code that makes as many calls as it can, but goes to sleep when it gets "would block", and tries again on SIGIO. Don't write it as code that only does stuff when SIGIO happens, or assumes that because it got SIGIO something has happened.

We're not trying to be difficult here, we're doing all that is possible on the stacks we need to support.

We need to support underlying network interfaces with intensely variable event generation, and which potentially provide far too many or too few events, with limited ability to read state. We can't pass up information that we're not given by the underlying stack. On some systems we could have to resort to just issuing SIGIO to every socket on a timer event - most signals would be totally spurious, because nothing's actually happening. We can only guarantee that we can produce enough signals to allow non-blocking code to retry as efficiently as possible on each platform, but not a precise pattern. (The _pending flag is an attempt to try to tame such scatter-gun generation, limiting it to one SIGIO between each socket call).

We're in basically the same situation as you are - we would /like/ to provide precise callbacks, but we can't if the underlying stack doesn't give us detailed information. If it was possible for us to provide precise callbacks on top of an arbitrary stack that doesn't give that information, then it would also be possible for you to generate your own precise callbacks on top of NSAPI.

This approach contrasts to the PAL network code - I've seen that it claims to provide select(), but without the ability to portably make a select() call that returns correct information - some existing implementations pretty much just make up results, so any code using select() doesn't work portably.

Instead we're being honest by giving you a limited API than can be implemented and relied on to do the limited amount it claims on every platform, rather than giving you a more advanced API that can't be trusted to actually do what it claims. (It may be that working poll/select() appears in future for some stacks in NSAPI, but reliance on it would limit you to that subset of stacks.)

I'm assuming you've yet to run on any NSAPI stack other than lwIP? You should try the ESP8266 - that generates a huge number of spurious events if there are multiple applications using sockets, and will be very different from lwIP+NSAPI. What about Nanostack? Testing your code on other stacks sooner rather than later should make it less likely to be fazed by minor variations in lwIP behaviour.

I'm beginning to wonder if we should just drop SIGIO altogether and force applications to just retry 10 times a second. (Would save an awful lot of typing on my part.) If your code just retried on a timer, it would free you from any signal variation in underlying stacks. We've given you the best possible portable wake-up signal, but if you'd prefer constant stack-invariant polling, that would work, albeit inefficiently. You don't have to use SIGIO at all.

Or you could call your callback from a timer /in addition/ to the calls from SIGIO - periodic timer-based kicks can cover up problems with edge-based signals.

kjbracey-arm, 2017-03-07:
> And if the behavior from mbedOS is different from stack to stack, then user needs to be able to query on which stack he is working now, and adapt his code with branching logic for every stack differently

If you are unable to cope with variations in SIGIO timing, then don't use it. Just retry socket calls using a timer after you get EWOULDBLOCK.

Then you will not see any noticeable variance - the socket calls should not vary much between platforms.

("Much" being the operative word - there could still be variance in how much you might expect to be transferred in one go per non-blocking call, for example).

kjbracey-arm, 2017-03-07:
> But then i see less logic in abstracting network stacks by mbedOS, their functionality can be as well exposed directly to user.

The /defined/ behaviour of NSAPI does not change from stack to stack - the implementation of the API follows its documentation on every platform, and on every version.

(Okay, there are some legacy implementation glitches like connect() being blocking-only on some stacks - we're in no rush to correct that one, as we've seen existing code written assuming connect() ignores the blocking flag).

Implement code relying only on the defined behaviour of NSAPI, and it will work on all stacks, without modification. If some stack incorrectly implements the defined NSAPI behaviour, it can be fixed.

Implement code relying on /undefined/ behaviour of NSAPI, and it that code will not work on all stacks, and will be likely to break on successive versions of mbed OS, as is the case for your code.

The defined behaviour for SIGIO is narrow, but I assert that it's the best that can be achieved for a totally portable wakeup mechanism. It is entirely sufficient to write correct, portable, efficient code. Tightening the definition would rule out a number of potential network interfaces.

Really, the socket layer should generally be hidden behind transport APIs, like HTTP request libraries or the CoAP service. Those will hide typical users from the many fine details and pitfalls of networking. SIGIO is but one of those pitfalls. And those APIs can have much more precise defined behaviour, because they're entirely implemented locally, rather than being an abstraction onto something quite variable.

But the people working on the socket layer directly will have to be able to handle the many fine details and traps of getting network comms right - SIGIO variability is just one of the traps.

sg-, 2017-04-19:
@NirSonnenschein Can this be closed?

NirSonnenschein, 2017-04-19:
Yes, when changing the code to work based on Kevin's inputs it seems to function as expected.
the behavior is not necessarily intuitive so it might be a good idea to expand the documentation to make sure it is clearer to the users.

closed 2017-04-19 by NirSonnenschein