ARMmbed/mbed-os #3884: Need check whether a pointer is valid before make overflow test

kyliuxing, 2017-03-06:
I met a bug when i call pbuf_alloc(PBUF_RAW, len, **PBUF_POOL**);, it did not check whether the pointer is NULL before make overflow test.
Thanks!

lwip\src\core\lwip_memp.c
#if !MEMP_OVERFLOW_CHECK
do_memp_malloc_pool(const struct memp_desc *desc)
#else
do_memp_malloc_pool_fn(const struct memp_desc *desc, const char* file, const int line)
#endif
{
.......
  memp = *desc->tab;

#if MEMP_OVERFLOW_CHECK == 1
  **if (memp != NULL) {**
    memp_overflow_check_element_overflow(memp, desc);
    memp_overflow_check_element_underflow(memp, desc);
  **}**
#endif /* MEMP_OVERFLOW_CHECK */
#endif /* MEMP_MEM_MALLOC */

  if (memp != NULL) {
#if !MEMP_MEM_MALLOC
    *desc->tab = memp->next;
.........
}

0xc0170, 2017-03-06:
cc @geky @kjbracey-arm

geky, 2017-03-07:
@kjbracey-arm, is this actually an issue that should be raised against upstream lwip?

kjbracey-arm, 2017-03-08:
Seems to be, yes.

I also believe that MEMP_OVERFLOW_CHECK isn't on in our builds (it would be turned on if someone enabled LWIP_DEBUG), so it doesn't hit normal operations.

janjongboom, 2017-03-17:
@geky, has this bug been created?

closed 2018-10-02 by adbridge