ARMmbed/mbed-os #3885: NUMAKER_PFM_NUC472 mac address is set to software default

mazimkhan, 2017-03-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Blocker

NUMAKER_PFM_NUC472 board mac address is not read successfully from hw and it uses the default address: 02:00:AC:55:66:77. Also the socket connection on this board are not stable.

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2017-03-06:
@cyliangtw @ccli8

cyliangtw, 2017-03-06:
@mazimkhan 
There is no eeprom on NUC472 board, so it will use the same default address. We will seek the better way.
More, how to reproduce socket connection unstable issue on your board ?

mazimkhan, 2017-03-06:
Updated the steps to reproduce above. Essentially it is unstable while running example https://github.com/ARMmbed/mbed-os-example-tls/tree/master/tls-client
Most of the times it will fail to connect the server (developer.mbed.com). Also some times there are socket errors.

cyliangtw, 2017-03-09:
@mazimkhan
After implemented unique locally administered MAC address, I tested tls-client sample and never suffered failure of HTTPS connection. I attached the log for your reference. I will apply one PR based on [nuvoton branch](https://github.com/OpenNuvoton/mbed/tree/nuvoton) asap.
[tls_client.txt](https://github.com/ARMmbed/mbed-os/files/830184/tls_client.txt)

cyliangtw, 2017-03-15:
@mazimkhan
Applied PR<https://github.com/ARMmbed/mbed-os/pull/3918>
This PR has been passed all checks and got changes approved.

0xc0170, 2017-03-17:
@mazimkhan The fix referenced above - can you test it that it resolves this issue?

cyliangtw, 2017-03-17:
Yes, I tested ts_client and didn't suffer any connection issue over 10 runs, you can check the attached log file  ts_client.txt in the comment issued 8 days ago .

closed 2017-10-27 by ChiefBureaucraticOfficer