ARMmbed/mbed-os #3887: Design patterns and documentation for EventQueue

daniel-cesarini-tridonic-com, 2017-03-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type:Question
- Related issue: https://github.com/ARMmbed/mbed-os/issues/3873
- Priority: Major

## Question

** Design patterns and documentation for EventQueue **
Is there any further documentation (beyond the .h files and the existing .md) about intended design patterns to be used with EventQueues. 
For instance recurring cases such as: 
* exit external source interrupts;
* periodic system activities;
* posting delays for timing.

Furthermore, how should we handle different priorities (different threads?)?

pan-, 2017-03-06:
@daniel-cesarini-tridonic-com You can consult [this](https://developer.mbed.org/blog/entry/Simplify-your-code-with-mbed-events/) blog post and the documentation (https://docs.mbed.com/docs/mbed-os-handbook/en/5.1/concepts/events/) to learn more about the `EventQueue`. 

> Furthermore, how should we handle different priorities (different threads?)?

What do you mean by *different threads* ? EventQueue are orthogonal to thread mechanism.
Events are ordered in the EventQueue by *execution timestamp* but there is no ordering mechanism in place for events which have to be executed at the same timestamp.

daniel-cesarini-tridonic-com, 2017-03-06:
@pan- thank you for providing those pointers. I already checked them. 
We are currently developing a system with memory constrained devices, and need to tailor memory usage throughout the system itself. Furthermore I need to better understand the caveats of EventQueue (side effects, memory management, ...), and possible have them in an organized and systematic structure. 

Regarding the *different threads* I mean that the EventQueues can be scheduled by different threads (if using the RTOS layer with different priorities), and thus should be able to treat the different queues appropriately (according to the respective priority of the scheduling threads).

@MarceloSalazar @geky @pan-

pan-, 2017-03-06:
> Regarding the different threads I mean that the EventQueues can be scheduled by different threads (if using the RTOS layer with different priorities), and thus should be able to treat the different queues appropriately (according to the respective priority of the scheduling threads).

Events from an event queue are dispatched from a single thread at a time while they can be scheduled from multiple thread (single consumer multiple producer queue) but events queue are never tied directly to any threads. 

It is the user code responsibility to dispatch/dequeue an event queue in a given thread. 
Internally, the event queue is protected against concurrent access by a critical section and the mechanism used by producers to signal a new work and by the consumer to wait for work is a `Semaphore` object. 

Thread scheduling obey to the general rules set by the RTOS when those primitives are in play.

closed 2017-10-27 by ChiefBureaucraticOfficer