ARMmbed/mbed-os #3889: e2 studio exporter template needs updating for RZ_A1H device.

phyokyaw, 2017-03-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: `#1756`
- Priority: Major

---------------------------------------------------------------
Added missing entry in template for the exporter to **e2 studio** IDE. 

**Target**
RZ_A1H

**Toolchain:**
GCC_ARM

**Expected behavior**
Generated project should build using e2 studio.

**Actual behavior**
mbed_config.h is missing.

0xc0170, 2017-03-06:
cc @TomoYamanaka

TomoYamanaka, 2017-03-09:
@0xc0170 

This issue is a same cause with the following Issue, we are working together.
  https://github.com/ARMmbed/mbed-os/issues/3593

@phyokyaw is dealing with this issue. Do you mind tracing with here?

0xc0170, 2017-03-13:
@TomoYamanaka Thanks for pinpoint the earlier bug report. This is a duplicate. @phyokyaw please follow #3593 issue.

closed 2017-03-13 by 0xc0170

phyokyaw, 2017-03-14:
@TomoYamanaka Thanks for letting me know.  I'll create a pull request this week as part of #3593.