ARMmbed/mbed-os #3890: [Exporters] make_gcc_arm doesn't use definitions in the default profile

pan-, 2017-03-06:
### Description
- Type: Bug 
- Related issue: https://github.com/ARMmbed/mbed-os-example-ble/issues/61
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target** ALL

**Toolchain:** GCC_ARM

**mbed-cli version:**  1.0.0

**meed-os sha:** 3a27568a505bbc0bb8eeb973109d3d39ce823d1c

**Expected behavior** Binary resulting from and export with make_gcc_arm should be similar to a binary compiled with `mbed compile -t GCC_ARM`

**Actual behavior** The default toolchain profile (develop) is not used by the exporter and as a result, the optimization flag is set to `-O0` instead of `-Os`.

**Steps to reproduce**: see https://github.com/ARMmbed/mbed-os-example-ble/issues/61

pan-, 2017-03-06:
@theotherjimmy

theotherjimmy, 2017-03-06:
The profile is set to `debug`. you can change it with the `--profile develop` command line option.

theotherjimmy, 2017-03-06:
I am working on a PR that will have exporters use all built-in profiles. I think that should resolve this issue, when it happens.

pan-, 2017-03-07:
> The profile is set to debug. you can change it with the --profile develop command line option.

The default profile used during compilation by mbed-cli is `develop`, why it has to be different for this exporter ? Can it be documented somewhere (or align with `mbed compile`) ?

theotherjimmy, 2017-03-07:
It's not different for this exporter. It's different for all exporters.

sg-, 2017-03-08:
This has always been the intention. When exporting you are taking advantage of debug capabilities therefore, getting default setting that are debug friendly is the default.

We should add a note to the docs about exporters and close this issue.

MarceloSalazar, 2017-03-10:
Can someone please take an action and/or confirm the (doc) fix / PR for this?

pan-, 2017-03-10:
> We should add a note to the docs about exporters and close this issue.

As an example maintainer, I would really appreciate to have more details in the [handbook](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/third_party/) about the exporters behavior, differences with mbed-cli, IDE versions targeted and possible corner cases. 

With such information I can point users of my example to an always up to date, complete documentation rather by "patching" the example documentation with partial information which might be outdated the next release.

theotherjimmy, 2017-04-26:
@pan- I will add some documentation for exporters in general. When I do, I will close this issue.

theotherjimmy, 2017-04-26:
Documentation added. Please follow up discussion in https://github.com/ARMmbed/Handbook/pull/151.

closed 2017-04-26 by theotherjimmy