ARMmbed/mbed-os #3891: lwip shouldn't use rand() for port or sequence number randomisation (or at all in general)

kuggenhoffen, 2017-03-06:
### Description
- Type: Enhancement
- Priority: Major

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
When using an application using TCP socket, I expect port numbering to start from different port between resets. Currently TCP socket ports always start from same number. This is because lwip uses rand() and srand is initialized to 1 by default in standard library. This results sometimes in unexpected behaviour when using 0 as local port and resetting the device since the previous TCP stream is left open and new one is attempted to open with same port number, depending on network and sequence numbers this might be ok or it might result in connect failing.

However as a user of mbedOS TCPSocket it is not obvious that if I pass in 0 as local port that I also would have to call srand with a random value to get different initial port. Even if a user would know to call srand, they would also have to know that no library is also calling srand with some constant value before the port is randomized.

**Suggested enhancement**
Use another method of random number generation within lwip, for example randLIB, so that behaviour is not dependent on application or other libraries.

0xc0170, 2017-03-06:
cc @geky @kjbracey-arm

kjbracey-arm, 2017-03-06:
Can't argue with this.  rand() and srand() feel to me like reserved-for-application API, and the C standard states that the "implementation" (ie C library) doesn't call either.

System components like network stacks feel to me like they should behave similarly - neither rely on an application seeding srand(), nor interfere with an application that wants to seed rand() manually for its own purposes.

randLIB in FEATURE_COMMON_PAL deals quite nicely with seeding issues (it hooks into mbed TLS entropy if available, and allows more entropy to be easily mixed in) as well as being a high-quality PRNG. It's why Nanostack doesn't (any longer) suffer similar problems.

On a closely-related note - latest lwIP also now has a hook to generate TCP initial sequence numbers, which are currently very similar after every boot. If improving rand() we should also bring that hook into use to fix the remaining part of the TCP-after-reboot problem.

geky, 2017-03-06:
This is reasonable to me, relying on rand just has practical issues in this situation.

@0xc0170, do you know if the trng API would be reasonable here? That would be an easy requirement for truly randomized ports.

kjbracey-arm, 2017-03-08:
TRNG device is slight overkill, and has the limitation that it's "monolithic" - a single on-board source that may or may not be present.

randLIB is a kind of hybrid in that it starts with the TRNG (via mbedtls_hardware_poll) if present, but now also provides hooks for third parties to chip in with extra randomness (much like mbedtls itself does). That allows radio drivers to add radio noise, or the network stack to add MAC addresses, for example.

The fact that you can have MAC (or RTC?) randomisation is better than nothing on a system with no real entropy source, and we don't absolutely require the true randomness that mbed TLS demands for crypto security.

I think making non-crypto things rely on TRNG for correct operation would just encourage people to come up with flakey non-crypto-standard implementations. It's hard enough getting people to come up with good TRNG when it is for the crypto. But we do need something better than apps doing srand(time()).

mikaleppanen, 2017-03-28:
Using existing random library sounds like a good solution. I'll start to make needed changes for this.

mikaleppanen, 2017-05-24:
Correction is now merged to master: https://github.com/ARMmbed/mbed-os/pull/4284

closed 2017-05-24 by 0xc0170