ARMmbed/mbed-os #3896: I2C constructor failure on K64F for PinMap pair PTD3/PTA14

mray19027, 2017-03-06:
### Description

Creating an I2C instance using SDA/SCL pins PTD3/PTA14 fails on K64F

This SDA/SCL pairing is specifically stated in the `PeripheralPins.c` file generated for the K64F target. In addition, in `PeripheralPins.c`, the `PinMap` arrays for SDA and SCL have differing sizes which causes a `PinMap mismatch`:

PinMap code:
```
/************I2C***************/
const PinMap PinMap_I2C_SDA[] = {
    {PTE25, I2C_0, 5},
    {PTB1 , I2C_0, 2},
    {PTB3 , I2C_0, 2},
    {PTC11, I2C_1, 2},
    {PTA13, I2C_2, 5},
    {PTD3 , I2C_0, 7},
    {PTE0 , I2C_1, 6},
    {NC   , NC   , 0}
};

const PinMap PinMap_I2C_SCL[] = {
    {PTE24, I2C_0, 5},
    {PTB0 , I2C_0, 2},
    {PTB2 , I2C_0, 2},
    {PTC10, I2C_1, 2},
    {PTA12, I2C_2, 5},
    {PTA14, I2C_2, 5},
    {PTD2 , I2C_0, 7},
    {PTE1 , I2C_1, 6},
    {NC   , NC   , 0}
};
```

Code to replicate error:
```
#include "mbed.h"
#include "cmsis.h"
#include "pinmap.h"
#include "PeripheralPins.h"

int main() {
    PinName pin_sda = PTD3;
    PinName pin_scl = PTA14;
    printf("SDA pin %d, SCL pin %d\r\n", pin_sda, pin_scl);

    I2C i2c(pin_sda, pin_scl);

    printf("This will never run because the I2C constructor will fail for this SDA/SCL pairing\r\n");
}
```

### Details

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 

**mbed-cli version:** 
1.0.0

**meed-os sha:**
e435a07d9

@maclobdell @BlackstoneEngineering

0xc0170, 2017-03-07:
Is it because of different size? I would expect lookup go through all (pins might not be the same number for those 2 pins (SDA and SCL)). 

@mray19027 Do you suspect an error is in generic layer or specific to K64F MCU?

cc @mmahadevan108

mray19027, 2017-03-07:
@0xc0170 Updated the issue and issue title.

This is a specific issue with the K64F MCU

BlackstoneEngineering, 2017-03-07:
@maclobdell

mray19027, 2017-03-07:
Potentially can be an error in the generic layer because I found another related to the SCL pin for I2C. See #3904

mray19027, 2017-03-07:
PTA14 is on a different HW block than PTD3. Therefore this should fail and is not an error. Closing the issue

closed 2017-03-07 by mray19027