ARMmbed/mbed-os #3901: Support for B-L072Z-LRWAN1

andcor02, 2017-03-07:
Just a quick question when support for the B-L072Z-LRWAN1 board will be adde

http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-discovery-kits/b-l072z-lrwan1.html

Users are receiving boards, HTML directs them to the page, but when compiling they receive.

"generic_placeholder not recognised"

Any thoughts?

0xc0170, 2017-03-07:
cc @bcostm @adustm

adustm, 2017-03-07:
Hello @andcor02 
ST plans it for Q2'17. But users could create the files following the tutorial here : https://developer.mbed.org/teams/ST/wiki/steps-to-create-a-new-STM32-platform
and we would review their pull request.

The web page was updated yesterday here https://developer.mbed.org/platforms/ST-Discovery-LRWAN1/

Kind regards

bcostm, 2017-05-09:
PR #4256 for this platform is merged. Can you please close this "issue"?

closed 2017-05-09 by 0xc0170