ARMmbed/mbed-os #3903: cmsis5: Issues with SiLabs Giant Gecko

bulislaw, 2017-03-07:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
EFM32GG_STK3700

**Toolchain:**
GCC & IAR

**Toolchain version:**
GCC 6.3.1
IAR 7.8

During the upgrade of CMSIS and RTX to versions 5 and 2 we noticed tests failing on Giant Gecko board. The update is not yet finished, but most of the other vendor's boards pass the tests. We are planning to merge the update with master in next couple of weeks. The code can be found on feature branch: https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5

`mbed test -m EFM32GG_STK3700 -t <compiler`

Test mostly pass while using ARM, there are two failing tests the same as when using master:
```
| EFM32GG_STK3700-ARM | EFM32GG_STK3700 | tests-mbed_drivers-lp_timeout                                                | FAIL    | 18.27              | shell       |
| EFM32GG_STK3700-ARM | EFM32GG_STK3700 | tests-mbed_hal-lp_ticker                                                     | FAIL    | 17.96              | shell       |
```

For GCC:
```
+-------------------------+-----------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                  | platform_name   | test suite                                                                   | result  | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT | 67.67              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-basic_test_default                | TIMEOUT | 67.74              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-case_async_validate               | TIMEOUT | 67.6               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-case_control_async                | TIMEOUT | 67.73              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-case_control_repeat               | TIMEOUT | 67.58              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-case_selection                    | TIMEOUT | 67.87              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-case_setup_failure                | TIMEOUT | 67.59              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | TIMEOUT | 67.59              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-control_type                      | TIMEOUT | 68.17              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | TIMEOUT | 68.1               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 9.95               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | TIMEOUT | 69.8               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT | 67.93              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-frameworks-utest-tests-unit_tests-test_setup_failure                | TIMEOUT | 68.31              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-tests-filesystem-heap_block_device                                  | TIMEOUT | 67.65              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | features-tests-filesystem-util_block_device                                  | TIMEOUT | 68.16              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | rtos-rtx2-target_cortex_m-tests-memory-heap_and_stack                        | OK      | 9.77               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-events-queue                                                           | TIMEOUT | 68.02              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-events-timing                                                          | TIMEOUT | 68.15              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-integration-basic                                                      | OK      | 9.54               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-c_strings                                                 | TIMEOUT | 68.68              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-dev_null                                                  | OK      | 12.17              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-echo                                                      | TIMEOUT | 67.62              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-generic_tests                                             | TIMEOUT | 67.83              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-lp_timeout                                                | TIMEOUT | 67.71              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-race_test                                                 | TIMEOUT | 67.71              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-rtc                                                       | TIMEOUT | 68.01              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-stl_features                                              | TIMEOUT | 67.82              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-ticker                                                    | TIMEOUT | 67.53              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-timeout                                                   | TIMEOUT | 67.73              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-wait_us                                                   | TIMEOUT | 67.61              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_functional-callback                                               | TIMEOUT | 67.88              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_functional-callback_big                                           | TIMEOUT | 67.96              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_functional-callback_small                                         | TIMEOUT | 67.82              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_functional-functionpointer                                        | TIMEOUT | 67.5               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-lp_ticker                                                     | TIMEOUT | 67.8               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-mbed-attributes                                              | TIMEOUT | 67.61              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-mbed-call_before_main                                        | OK      | 13.17              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-mbed-cpp                                                     | OK      | 10.0               | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-mbed-div                                                     | OK      | 10.02              | shell       |
[... cut ...]
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedtls-selftest                                                       | TIMEOUT | 68.16              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-netsocket-ip_parsing                                                   | TIMEOUT | 67.92              | shell       |
+-------------------------+-----------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
```

For IAR all the tests time out.

@0xc0170 could you please tag SiLabs guys so they can help debugging the board.

0xc0170, 2017-03-08:
cc @stevew817

stevew817, 2017-03-08:
Would be nice to get some more information other than 'TIMEOUT'. Test output, for example, or compiler warnings. Did you test other Silicon Labs boards? EFM32PG, for example, which is the one we added to CI during the last partner workshop?
My guess is this has little to do with the actual CMSIS upgrade, since I'd then expect all tests to fail instead of 3/4. Everything going red on IAR does sound like an incompatibility somewhere.
Possibly a CI issue?
I can take a look, but it won't be very soon with EmbeddedWorld around the corner.

bulislaw, 2017-03-08:
I'm afraid I don't have much more details as the tests just hang. I'll try to connect debugger to get some more details. I would assume it's issues with the tests, but they pass on most of the platforms and they used to pass on GG before the CMSIS/RTX upgrade. I don't have PG, but I'll try to find one, we should have one.

bulislaw, 2017-03-10:
@stevew817 the issues looks extremely similar to #3683 I have issues setting up debugger (Simplicity studio is able to track asm but doesn't tie it with C code for some reasons) but at some point there's a TIMER2_IRQ being risen and the execution is stuck in default handler. It sounds exactly like what was happening for ST, it seems that with the RTX update the bootup sequence/timing changed slightly causing issues with some boards, they proposed fix (for their boards) in #3914 I was trying to have a look at how it's done in `mbed_sdk_init` and `chip_init` but without knowing the board is a bit hard to figure out what's going on there. Any chance you could have a look at that?

stevew817, 2017-03-16:
This isn't a timing issue. The vector relocation is simply not happening, because at some point you guys removed our cmsis_nvic.c file which was responsible for relocating the vector table to RAM upon the first call to NVIC_SetVector.
In CMSIS5, where CMSIS provides the NVIC_SetVector function, who is responsible for copying the vector table and setting VTOR accordingly?

bulislaw, 2017-03-16:
@stevew817 it's done here https://github.com/ARMmbed/mbed-os/blob/feature_cmsis5/rtos/mbed_boot.c#L273 I can't exactly remember why this function is called so late in the boot sequence, but I've also tried moving it as first thing we do after getting execution in mbed, without positive results.

stevew817, 2017-03-16:
@bulislaw Thanks for the info. So far, what I can see in debugging is that TIMER0 (which is the us_ticker timer) fires an interrupt without the interrupt table having been relocated (i.e. VTOR is still at 0, and the vector is pointing to the default handler as you mentioned).
Maybe worth investigating why the relocation seems to happen on ARMCC and not on GCC?

bulislaw, 2017-03-16:
Thanks for the hint! I'll try to confirm what's going on with the relocation.

stevew817, 2017-03-16:
@bulislaw I found the issue, GCC tests are passing now. See PR in #3944

bulislaw, 2017-03-16:
Thanks for the fix. We've fixed (mostly) IAR at the head of the branch and I'm getting the same kind of errors as we could see for gcc. I'll try to spend some time tomorrow to figure out whether it's similar issue with the vectors not being copied for some reason.

stevew817, 2017-03-17:
@bulislaw From what I can see when debugging IAR, the thread that is supposed to run pre_main never gets scheduled (`osThreadNew((osThreadFunc_t)pre_main, NULL, &_main_thread_attr)` returns 0)
Since that function calls straight into RTX2, I can't really debug efficiently any further at this point without going down a road of assumptions.

0xc0170, 2017-03-20:
> @bulislaw From what I can see when debugging IAR, the thread that is supposed to run pre_main never gets scheduled (osThreadNew((osThreadFunc_t)pre_main, NULL, &_main_thread_attr) returns 0)
Since that function calls straight into RTX2, I can't really debug efficiently any further at this point without going down a road of assumptions.

I'll try to reproduce this locally this week, or we do it the next week.

stevew817, 2017-03-29:
@0xc0170 Did you ever get around to this?

0xc0170, 2017-03-29:
> @0xc0170 Did you ever get around to this?

I should get a board from you tomorrow morning, and will test

0xc0170, 2017-03-30:
I've just run tests locally, latest feature_cmsis5:

``mbed test -n tests-mbedmicro-rtos-* -m EFM32GG_STK3700 -t IAR``

```
+---------------------+-----------------+-------------------------------------+---------+--------------------+-------------+
| target              | platform_name   | test suite                          | result  | elapsed_time (sec) | copy_method |
+---------------------+-----------------+-------------------------------------+---------+--------------------+-------------+
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-basic     | OK      | 20.9               | shell       |
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-isr       | TIMEOUT | 30.61              | shell       |
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-malloc    | OK      | 25.44              | shell       |
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-mutex     | OK      | 20.58              | shell       |
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-queue     | OK      | 12.97              | shell       |
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-semaphore | OK      | 17.71              | shell       |
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-signals   | OK      | 15.38              | shell       |
| EFM32GG_STK3700-IAR | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-threads   | OK      | 11.82              | shell       |
+---------------------+-----------------+-------------------------------------+---------+--------------------+-------------+
```

only isr test failes, will check what is going on

0xc0170, 2017-03-30:
Giant gecko, GCC ARM:

```
+-------------------------+-----------------+-------------------------------------+---------+--------------------+-------------+
| target                  | platform_name   | test suite                          | result  | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+-------------------------------------+---------+--------------------+-------------+
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-basic     | FAIL    | 25.37              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-isr       | OK      | 15.27              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-malloc    | OK      | 25.01              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-mutex     | OK      | 20.61              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-queue     | OK      | 11.93              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-semaphore | OK      | 17.77              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-signals   | OK      | 15.33              | shell       |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbedmicro-rtos-mbed-threads   | TIMEOUT | 50.28              | shell       |
+-------------------------+-----------------+-------------------------------------+---------+--------------------+-------------+
```

stevew817, 2017-12-04:
@bulislaw is this still an issue?

bulislaw, 2017-12-05:
The tests are passing on the master. Closing.

closed 2017-12-05 by bulislaw